platformio/platformio-core-installer #539: Installation Manager

wjptony, 2021-02-26:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.50.1
  PIO IDE: v2.2.1
  System: Windows_NT, 6.3.9600, x64

  %23 Exception
  ```
  TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received type undefined
	at validateString (internal/validators.js:112:11)
	at Object.dirname (path.js:583:5)
	at d.value (c:\Users\Tony\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:872685)
	at processTicksAndRejections (internal/process/task_queues.js:94:5)
	at async d.value (c:\Users\Tony\.vscode\extensions\platformio.platformio-ide-2.2.1\node_modules\platformio-node-helpers\dist\index.js:1:870529)
	at async t.default.install (c:\Users\Tony\.vscode\extensions\platformio.platformio-ide-2.2.1\dist\extension.js:1:6521)
	at async c:\Users\Tony\.vscode\extensions\platformio.platformio-ide-2.2.1\dist\extension.js:1:12506
  ```

ivankravets, 2021-03-22:
Have you resolved this issue?

ivankravets, 2021-03-23:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/533

closed 2021-03-23 by ivankravets