platformio/platformio-core-installer #540: PlatformIO installer messes up sys.path

Psirus, 2021-02-27:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Ubuntu 20.04

**PlatformIO Version**): PlatformIO Core, version 5.1.0

### Description of problem

I've ran the installer as per the docs. (`python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"`) Now, several directories are missing from sys.path and modules that I've installed via pip are not being found.

#### Steps to Reproduce

1. Have a setup without PlatformIO installed. Install a pip packages, say `pyyaml` and try `import yaml`. Works.
2. Install PlatformIO.
3. ModuleNotFoundError: No module named 'yaml'.

### Actual Results

The sys.path in IPython (maybe this is a clue, the path in IPython is not effected):

~~~
['/usr/bin',
 '/usr/lib/python38.zip',
 '/usr/lib/python3.8',
 '/usr/lib/python3.8/lib-dynload',
 '',
 '/home/psirus/.local/lib/python3.8/site-packages',
 '/usr/local/lib/python3.8/dist-packages',
 '/usr/lib/python3/dist-packages',
 '/usr/lib/python3/dist-packages/IPython/extensions',
 '/home/psirus/.ipython']
~~~

The sys.path in Python itself:

~~~
['', 
'/usr/lib/python38.zip', 
'/usr/lib/python3.8',
'/usr/lib/python3.8/lib-dynload',
'/home/psirus/.platformio/penv/lib/python3.8/site-packages']
~~~
 
Note in particular  '/usr/local/lib/python3.8/dist-packages' and '/usr/lib/python3/dist-packages' are missing.

### Expected Results

I probably expected the installer to append to the path, but not remove entries.

### Additional Information

My `PYTHONPATH` variable is empty.

ivankravets, 2021-02-27:
Sorry, we don't modify PYTHONPATH. You can check the installer source code.

closed 2021-02-27 by ivankravets

Psirus, 2021-02-27:
Okay, I've checked, and you're right, the installer did not modify the path. However, when I added `~/.platformio/penv/bin` to my `PATH`, the python executable in there exhibits this modified behavior compared to the system python. I didn't notice that I was calling this symlinked python executable. So maybe not a bug, but still unexpected, at least for me.