platformio/platformio-core-installer #553: Problème installation platformIO

wintternox, 2021-03-05:
Bonjour à tous, 
J'ai un soucis, je n'arrive pas à installer platformIO sur VS code. Quand je lance l'installation depuis VS code j'ai ce message d'erreur: 
[image]
pouvez vous m'aidez??  Je suis trop dans la galère sinon 

merci par avance ;)

dsferruzza, 2021-03-06:
Même problème pour moi ; j'ai rollback sur la version précédente de l'extension (Ctrl + Shift + P, puis `Extensions: Install Specific Version of Extension`) en attendant

kergaradec, 2021-03-08:
J'ai également le même soucis et le retour en version précédente n'a eu aucun effet, donc mon problème persiste.

ouistiti91, 2021-03-22:
Même problème, résolu en installant python avant d'installer platformIO. Dans mon cas, je suis passé par microsoft store pour installer python 3.9.

closed 2021-04-29 by stale[bot]