platformio/platformio-core-installer #569: Windows store python 3.9 not detected

arbabseyfola, 2021-03-07:
Plafromio not detect window store python.

ivankravets, 2021-03-07:
We ignore Python distribution from the Microsoft store. It is broken. We can't create a Python virtual environment from it. 

PlatformIO IDE comes with a built-in portable Python located in `%USERPROFILE/%.platformio/python3`. Could try to run?

```
`C:/Users/XXX-REPLACE-WITH-YOUR-USERNAME-XXX/.platformio/python3/python --version
```

See related comment https://github.com/platformio/platformio-core-installer/issues/221#issuecomment-792263689

SebastianBuslon, 2021-03-07:
Hi, I've the same problem, I have read everything and I didn't get resolved.  When I try to add the path I only have the following path

`C:/Users/XXX-REPLACE-WITH-YOUR-USERNAME-XXX/.platformio/python3/

with out /python --version,   I have this:

C:\Users\Sebas\.platformio\python3\python.exe.

I added it anyway and it isn't work.

And the installer still look for python in Mocrosoft  app folder.

Error: Traceback (most recent call last): File "C:\.platformio\.cache\tmp\.piocore-installer-hz4irpd2\tmpjorvxir3\pioinstaller.zip\pioinstaller\python.py", line 170, in find_compatible_pythons File "C:\.platformio\python3\lib\subprocess.py", line 420, in check_output return run(*popenargs, stdout=PIPE, timeout=timeout, check=True, File "C:\.platformio\python3\lib\subprocess.py", line 524, in run raise CalledProcessError(retcode, process.args, subprocess.CalledProcessError: Command '['C:\\Users\\Sebas\\AppData\\Local\\Microsoft\\WindowsApps\\python3.exe', 'C:\\.platformio\\.cache\\tmp\\get-platformio-1.0.0.py', '--no-shutdown-piohome', 'check', 'python']' returned non-zero...

Could you help me?

BR
Sebastian

closed 2021-04-15 by stale[bot]

mapj1984, 2021-06-27:
this work for me:

https://github.com/platformio/platformio-core-installer/issues/221

Solution 1
Press Windows
Start typing: App Execution Aliases
Disable Python and Python3
Restart VSCode.