platformio/platformio-core-installer #576: cant install platformio ide in vscode

sarberto, 2021-03-09:
Hi, would somebody help me? I´m novice in this field, only need platformio to change my 3d printer marlin.

try many times but always same text

Error: Traceback (most recent call last):
  File "C:\Users\usuaio\.platformio\.cache\tmp\.piocore-installer-h_nl0xsx\tmp2bsckmq5\pioinstaller.zip\pioinstaller\python.py", line 170, in find_compatible_pythons
  File "C:\Users\usuaio\.platformio\python3\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\usuaio\.platformio\python3\lib\subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\Users\\usuaio\\AppData\\Local\\Microsoft\\WindowsApps\\python.exe', 'C:\\Users\\usuaio\\.platformio\\.cache\\tmp\\get-platformio-1.0.0.py', '--no-sh...

i tried every tipical solution and others i see here and there: changing path to python39, uninstall avast, disable windows firewall, of course create .platformio folder in users and c: ,execute getplatformio in cmd and folder, nothing change when i open vscode install starts and ends in error.
I´m using windows 10.

bbyku, 2021-03-10:
Hi 
I have same issue.It helped when i was inside vscode in TERMINAL just type PYTHON it should open windows store and install it from there then restart VSCODE and should work
GOOD LUCK

sarberto, 2021-03-10:
hi, thanks for your advise, i try but nothing happened , i have already python 3.9 installed before doing that, messagge in terminal told me version but it still return error with paltformio installation.

bbyku, 2021-03-10:
just try it in terminal but the one inside vscode or go to microsoft store I had also instaled it alone and didnt work through visual code and microsoft store should work

sarberto, 2021-03-11:
try both ways, still same error :(

AlemarFernandes, 2021-03-27:
estou com este mesmo problema nao abre , ja usei uma vez mas agora nao esta abrindo

closed 2021-05-05 by stale[bot]