platformio/platformio-core-installer #629: Installation Manager

saltek2005, 2021-03-20:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.3
  PIO IDE: v2.3.0
  System: Windows_NT, 6.1.7601, ia32

  %23 Exception
  ```
  Error: Can not find Python Interpreter. Please install Python 3.6 or above manually
	at E.value (c:\Users\Clear\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:924721)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
	at async E.value (c:\Users\Clear\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:923268)
	at async t.default.install (c:\Users\Clear\.vscode\extensions\platformio.platformio-ide-2.3.0\dist\extension.js:1:6509)
	at async c:\Users\Clear\.vscode\extensions\platformio.platformio-ide-2.3.0\dist\extension.js:1:12420
  ```

ivankravets, 2021-03-22:
Do you use 3rd party antivirus tools?

abdset, 2021-04-06:
Did you checked Add path when installing python ?

closed 2021-05-14 by stale[bot]