platformio/platformio-core-installer #635: Installation Manager

vx4500, 2021-03-21:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.53.2
  PIO IDE: v2.3.0
  System: Windows_NT, 6.3.9600, x64

  %23 Exception
  ```
  TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received undefined
    at validateString (internal/validators.js:120:11)
    at Object.pythonPrompt [as dirname] (path.js:583:5)
    at E.value (c:\Users\vx\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\installer\stages\platformio-core.js:179:37)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at E.value (c:\Users\vx\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\installer\stages\platformio-core.js:104:20)
    at t.default.install (c:\Users\vx\.vscode\extensions\platformio.platformio-ide-2.3.0\dist\webpack:\platformio-ide\src\installer\manager.js:86:57)
    at c:\Users\vx\.vscode\extensions\platformio.platformio-ide-2.3.0\dist\webpack:\platformio-ide\src\main.js:207:36
  ```

ivankravets, 2021-03-22:
Have you resolved this issue?

ivankravets, 2021-03-23:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/533

closed 2021-03-23 by ivankravets