platformio/platformio-core-installer #641: Installation Manager

Vcelist, 2021-03-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.3
  PIO IDE: v2.3.0
  System: Windows_NT, 6.1.7601, ia32

  %23 Exception
  ```
  TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received undefined
	at validateString (internal/validators.js:120:11)
	at Object.dirname (path.js:583:5)
	at E.value (c:\Users\V\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:924600)
	at processTicksAndRejections (internal/process/task_queues.js:97:5)
	at async E.value (c:\Users\V\.vscode\extensions\platformio.platformio-ide-2.3.0\node_modules\platformio-node-helpers\dist\index.js:1:923268)
	at async t.default.install (c:\Users\V\.vscode\extensions\platformio.platformio-ide-2.3.0\dist\extension.js:1:6509)
	at async c:\Users\V\.vscode\extensions\platformio.platformio-ide-2.3.0\dist\extension.js:1:12420
  ```

ivankravets, 2021-03-22:
Have you resolved this issue?

ivankravets, 2021-03-23:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/533

closed 2021-03-23 by ivankravets