platformio/platformio-core-installer #674: Installation Manager

Andrew55529, 2021-04-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.55.0
  PIO IDE: v2.3.1
  System: Windows_NT, 10.0.19041, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.0.1
Platform: Windows-10
Python version: 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]
Python path: C:\Users\Andrey\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Andrey\.platformio\penv

	at c:\Users\Andrey\.vscode\extensions\platformio.platformio-ide-2.3.1\node_modules\platformio-node-helpers\dist\index.js:1:931162
	at ChildProcess.o (c:\Users\Andrey\.vscode\extensions\platformio.platformio-ide-2.3.1\node_modules\platformio-node-helpers\dist\index.js:1:931056)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:483:12)
	at ChildProcess.cp.emit (c:\Users\Andrey\.vscode\extensions\platformio.platformio-ide-2.3.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
  ```

derekpiasecki, 2021-04-05:
I am hitting this same issue.  Some more details:  I newly (re)installed VSCode after wiping a computer, and when complete it automatically installed the extensions I had previously, including Platform.io.  Platform.io then proceeded to install its various subcomponents, and while installing Core, it failed with this same exception.  I have tried restarting VSCode and letting it attempt to continue installing multiple times.

ivankravets, 2021-04-06:
@derekpiasecki do you use 3rd party antivirus tools?

derekpiasecki, 2021-04-06:
No, just MS's.

Andrew55529, 2021-04-06:
I delete platformio and rename .platformio to another name.
Then I install platforming. And it's start working.

derekpiasecki, 2021-04-06:
Results of testing installation of older versions show all 2.x.y versions have this error (except 2.1.1 which fails differently - it doesn't even show up or attempt to install Core).  I didn't test multiple 1.x.y versions, but 1.1.0 installed without errors.  Note that I had previously installed (a few months ago) something in the 2.x.y series successfully, but don't know the exact version.

derekpiasecki, 2021-04-06:
Based on @Andrew55529 's suggestion, I tried narrowing things down a bit.  First I deleted the .platformio/.cache folder, but that didn't help.  I then tried deleting both .platformio/python* subfolders, and that fixed the problem.  Also note that after the successful install, I had only the one .platformio/python3 folder.

closed 2021-05-14 by stale[bot]