platformio/platformio-core-installer #975: Installation Manager

jwouda, 2021-07-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  -  

  %23 Configuration

  VSCode: 1.58.2
  PIO IDE: v2.3.2
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\jeroe\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 69, in <module>
    main()
  File "C:\Users\jeroe\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 61, in main
    bootstrap()
  File "C:\Users\jeroe\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\pioinstaller\__main__.py", line 131, in main
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\pioinstaller\core.py", line 67, in install_platformio_core
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\pioinstaller\core.py", line 95, in _install_platformio_core
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\Users\jeroe\.platformio\.cache\tmp\.piocore-installer-47t_yik1\tmp71k5g9d7\pioinstaller.zip\pioinstaller\python.py", line 199, in find_compatible_pythons
  File "C:\Users\jeroe\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\jeroe\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\jeroe\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\jeroe\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 1] Incorrect function
 -> Installer version: 1.0.2
Platform: Windows-10
Python version: 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]
Python path: C:\Users\jeroe\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\jeroe\.platformio\penv

	at c:\Users\jeroe\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\index.js:1:931376
	at ChildProcess.a (c:\Users\jeroe\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\index.js:1:931270)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\jeroe\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2021-08-14:
Please update the PlatformIO IDE extension in the VSCode to the latest version. The bug is fixed.

closed 2021-08-14 by ivankravets

Ruben22222, 2021-08-16:
Installing PlatformIO IDE...
It may take a few minutes depending on your connection speed
Please do not close this window and do not open other folders until this process is completed.

Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
Failed to install PlatformIO IDE.
formatie hasta el pc, no hay forma de instalar platform

Ruben22222, 2021-08-16:
Error: Could not start PIO Home server: Error: timeout
	at c:\Users\Dinamo\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:5946
	at runNextTicks (internal/process/task_queues.js:58:5)
	at processTimers (internal/timers.js:494:9)...