platformio/platformio-core-installer #978: Installation Manager Not compliant with AV

JeanMRog, 2021-07-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.3
  PIO IDE: v2.3.1
  System: Windows_NT, 10.0.18363, x64

  %23 Exception
  ```
  Error: Command "python setup.py egg_info" failed with error code 1 in C:\Users\to85962\.platformio\.cache\tmp\.piocore-installer-l5c20znq\pip-install-vrcqgmyv\platformio\
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\to85962\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.0.1
Platform: Windows-10
Python version: 3.6.8 (tags/v3.6.8:3c6b436a57, Dec 24 2018, 00:16:47) [MSC v.1916 64 bit (AMD64)]
Python path: C:\Program Files\Simcenter\2020.2\Amesim\sys\python\win64\python.exe
Creating a virtual environment at C:\Users\to85962\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Looking in indexes: https://artifactory.2b82.aws.cloud.airbus.corp/artifactory/api/pypi/r-airbus-pypi-virtual/simple, https://artifactory.fr.eu.airbus.corp/1Y40/api/pypi/R-E32B_HiveOps_pypi_L/simple, https://artifactory.fr.eu.airbus.corp/1Y40/api/pypi/pypi/simple, https://artifactory.fr.eu.airbus.corp/1Y40/api/pypi/R-2H89_analytics-community_pypi_L/simple, https://artifactory.airbus.corp/artifactory/api/pypi/r-2h89-analytics-community-pypi-local/simple
Collecting platformio
  Downloading https://artifactory.2b82.aws.cloud.airbus.corp/artifactory/api/pypi/r-airbus-pypi-virtual/packages/packages/b8/f0/4e9941349c3450838667d323545b256d91b723a4685a1e056f01642b851e/platformio-5.1.1.tar.gz (215kB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\to85962\.platformio\penv\lib\site-packages\setuptools\__init__.py", line 11, in <module>
    ModuleNotFoundError: No module named 'setuptools._deprecation_warning'
    
    ----------------------------------------

	at c:\Users\to85962\.vscode\extensions\platformio.platformio-ide-2.3.1\node_modules\platformio-node-helpers\dist\index.js:1:931162
	at ChildProcess.o (c:\Users\to85962\.vscode\extensions\platformio.platformio-ide-2.3.1\node_modules\platformio-node-helpers\dist\index.js:1:931056)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:483:12)
	at ChildProcess.cp.emit (c:\Users\to85962\.vscode\extensions\platformio.platformio-ide-2.3.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
  ```

ivankravets, 2021-08-05:
What is "AV"? Do you mean "antivirus"?

JeanMRog, 2021-09-05:
I use Platform IO to program arduinos & like boards for prototyping
activities.

It is currently not working well.

what shall I do ?

*Dr Jean-Michel ROGERO*

*Test early... Physics always win at the end*

*Airbus Operations SAS*

Powerplant - Innovation & Data Sciences Manager

Mob:    +33 7 50 12 22 59

The information in this e-mail is confidential. The contents may not be disclosed or used by anyone other than the addressee. Access to this e-mail by anyone else is unauthorised.
If you are not the intended recipient, please notify Airbus immediately and delete this e-mail.
Airbus cannot accept any responsibility for the accuracy or completeness of this e-mail as it has been sent over public networks. If you have any concerns over the content of this message or its Accuracy or Integrity, please contact Airbus immediately.
All outgoing e-mails from Airbus are checked using regularly updated virus scanning software but you should take whatever measures you deem to be appropriate to ensure that this message and any attachments are virus free.

ivankravets, 2021-09-07:
@JeanMRog could you provide more details on what does not work correctly?

closed 2021-10-15 by stale[bot]