platformio/platformio-core-installer #999: Installation Manager

carolinamistretta, 2021-08-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.58.2
  PIO IDE: v2.3.2
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Usuario\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 69, in <module>
    main()
  File "C:\Users\Usuario\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 61, in main
    bootstrap()
  File "C:\Users\Usuario\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 45, in bootstrap
    import pioinstaller.__main__
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\Usuario\.platformio\.cache\tmp\.piocore-installer-qsi9btoj\tmphrenzc_c\pioinstaller.zip\pioinstaller\__main__.py", line 20, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\Usuario\.platformio\.cache\tmp\.piocore-installer-qsi9btoj\tmphrenzc_c\pioinstaller.zip\click\__init__.py", line 7, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\Usuario\.platformio\.cache\tmp\.piocore-installer-qsi9btoj\tmphrenzc_c\pioinstaller.zip\click\core.py", line 9, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\Usuario\.platformio\.cache\tmp\.piocore-installer-qsi9btoj\tmphrenzc_c\pioinstaller.zip\click\_compat.py", line 667, in <module>
  File "<frozen zipimport>", line 259, in load_module
  File "C:\Users\Usuario\.platformio\.cache\tmp\.piocore-installer-qsi9btoj\tmphrenzc_c\pioinstaller.zip\click\_winconsole.py", line 10, in <module>
  File "C:\Users\Usuario\.platformio\python3\lib\ctypes\__init__.py", line 8, in <module>
    from _ctypes import Union, Structure, Array
ModuleNotFoundError: No module named '_ctypes'

    at cmd (c:\Users\Usuario\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:188:25)
    at ChildProcess.a (c:\Users\Usuario\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:178:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\Usuario\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2021-08-05:
Do you use antivirus tools or firewalls?

r4wabr, 2021-08-05:
Disable Alias for python.

r4wabr, 2021-08-05:
OR run script manualy from Python IDE

ivankravets, 2021-08-06:
@r4wabr do you have any issues with this installer when "Alias for Python" is enabled?

r4wabr, 2021-08-09:
When "Alias for Python" is enabled  plagin do not installing. When disable - all installing ok.Василий, R4WABR.73!

ivankravets, 2021-08-09:
@r4wabr could you contact me directly via me@ikravets.com? I would be thankful if you help us to debug this issue.

closed 2021-09-15 by stale[bot]