platformio/platformio-core-installer #1021: Installation Manager

FallenAvatar, 2021-08-10:
%23 Description of problem
  Just tried installing the extension. Fully downloaded/"installed" in the VSCode Extension page, then started actually installing (bottom right messages) and the below happened. Not seeing any actual error/exception messages for me to search for unfortunately. Best guess I had from the traceback and buried error was the python.exe was corrupt, or wrong path. But I checked and it exists at the path mentioned and is able to start in interactive mode without issues.

Let me know if you need any more specific info.

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.59.0
  PIO IDE: v2.3.2
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Amy\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 69, in <module>
    main()
  File "C:\Users\Amy\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 61, in main
    bootstrap()
  File "C:\Users\Amy\.platformio\.cache\tmp\get-platformio-1.0.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\pioinstaller\__main__.py", line 131, in main
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\click\core.py", line 829, in __call__
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\click\core.py", line 782, in main
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\click\core.py", line 1236, in invoke
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\click\core.py", line 1066, in invoke
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\click\core.py", line 610, in invoke
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\click\decorators.py", line 21, in new_func
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\pioinstaller\core.py", line 67, in install_platformio_core
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\pioinstaller\core.py", line 95, in _install_platformio_core
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\pioinstaller\penv.py", line 52, in create_core_penv
  File "C:\Users\Amy\.platformio\.cache\tmp\.piocore-installer-g1dw_ook\tmpd4gl_o35\pioinstaller.zip\pioinstaller\python.py", line 199, in find_compatible_pythons
  File "C:\Users\Amy\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Amy\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Amy\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Amy\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 15612] The application cannot be started. Try reinstalling the application to fix the problem
 -> Installer version: 1.0.2
Platform: Windows-10
Python version: 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]
Python path: C:\Users\Amy\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Amy\.platformio\penv

    at cmd (c:\Users\Amy\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:188:25)
    at ChildProcess.a (c:\Users\Amy\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:178:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\Amy\.vscode\extensions\platformio.platformio-ide-2.3.2\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

FallenAvatar, 2021-08-10:
Closing VSCode, and re-opening caused PlatformIO to start installing again and it succeeded this time. One more restart at end of install as requested, and it appears to all be working.

ivankravets, 2021-08-14:
Please update the PlatformIO IDE extension in the VSCode to the latest version. The bug is fixed.

closed 2021-08-14 by ivankravets