platformio/platformio-core-installer #1038: Installation Manager

OzHawk, 2021-08-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.47.3
  PIO IDE: v2.3.3
  System: Linux, 4.15.0-36-generic, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "file" argument must be of type string. Received type object
	at /home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771390
	at a (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771284)
	at w (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771035)
	at u (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:770559)
	at /home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771343
	at new Promise (<anonymous>)
	at l (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771322)
	at e.callInstallerScript (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:7215)
	at async H.value (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:763673)
	at async t.default.install (/home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/dist/extension.js:1:6773)
	at async /home/mike/.vscode/extensions/platformio.platformio-ide-2.3.3/dist/extension.js:1:12776
  ```

JamesNewton, 2021-09-14:
You seem to be the first person reporting this problem... gobs of others have not bothered to search. Or failed to see this.

OzHawk, 2021-09-14:
Do you need any more information to replicate the issue?

JamesNewton, 2021-09-14:
Sorry if I wasn't clear... I'm not a dev here. I'm /also/ having the problem, I only commented because I wanted to follow the first case so I will hear if it's fixed. 

The workaround I've found is simply to install the prior version:
1. Click on Extensions (or View / Extensions)
2. Next to PlatformIO IDE, click on the little gear, then select "Install another version..."
3. Once that finishes, restart VS Code, let the new (old) version of PlatformIO finish installing, and restart VS Code again.

JamesNewton, 2021-09-14:
Oh! Fixed it. Problem is the default installed version of Python3. 

If you go to terminal and enter `python3 --version` and get anything before 3.6, that's the issue. PlatformIO stopped supporting 3.5 and better. Now, installing 3.9 is a nightmare and setting it as the default for python3 is also a pain. 

I have Ubuntu 16.04

`ll /usr/bin/python*` will show you what versions of python you actually have installed. 

To install the latest Python, what worked for me was:
```
sudo apt install software-properties-common
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt update
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt install python3.9 python3.9-venv python3.9-dev
```

And then `python3.9 -V` should show the version, but `python3 --version` will still show the old version. 

You are supposed to be able to see all the version with 
`sudo update-alternatives --config python3`
but it showed `update-alternatives: error: no alternatives for python`

So a bit more googling found that we can do:
`update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.9 2`
And then since I had 3.5 I also did
`sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.5 1`
setting it to the lower priority. Now 
`sudo update-alternatives --config python3`
works but it's already configured to 3.9

And... now PlatformIO installs the latest version! Woo Hoo!

JamesNewton, 2021-09-14:
Hey, do a bunch of people a favor and rename your issue to include [SOLVED] so other people will find it and see the solution?

OzHawk, 2021-09-14:
I'll check it out and make [SOLVED] if that was my issue also.

I'm running Ubuntu 20.04 which has a default Python version of 3.8, so
I'm not sure that it will fix my problem.

Cheers,
Mike

JamesNewton, 2021-09-16:
BTW, doing that change on the default python screwed up my OS... it wouldn't check for updates any more. Had to set it back to 3.5.

tenkoh2, 2021-10-04:
Changing the environment variable "PYTHONPATH" from "C: \ Python27 \ Lib" to "C: \ Users \ {UserName} \ AppData \ Local \ Programs \ Python \ Python39 \ Lib" resolved the error.

JamesNewton, 2021-10-05:
> Changing the environment variable "PYTHONPATH" from "C: \ Python27 \ Lib" to "C: \ Users \ {UserName} \ AppData \ Local \ Programs \ Python \ Python39 \ Lib" resolved the error.

@tenkoh2 That would be for a Windows OS? Would you share what version? 

On Ubuntu 16.04, if I change the default python version, the OS has issues (can't find updates, etc...) so it may be that PlatformIO no longer supports the older OS's. (16.04 is near EOL) and I may have to upgrade to 18 or even 20.