platformio/platformio-core-installer #1075: Installation Manager

xiaobaishangche, 2021-08-29:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.59.1
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390
	at a (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
	at w (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035)
	at u (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559)
	at c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771343
	at new Promise (<anonymous>)
	at l (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771322)
	at e.callInstallerScript (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:7215)
	at async H.value (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:763673)
	at async t.default.install (c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:6773)
	at async c:\Users\12425\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:12776
  ```

corvuslupus, 2021-10-02:
I'm seeing the same problem with VSC 1.60.2 on Windows 10.0.19042:

Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null at c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390 at a (c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284) at w (c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035) at u (c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559) at c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.3.3\node...

corvuslupus, 2021-10-02:
Nevermind... Just had to re-install Python.  It's installing correctly now.