platformio/platformio-core-installer #1158: Installation Manager

samandiriel, 2021-09-21:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.1
  PIO IDE: v2.3.3
  System: Linux, 4.15.0-142-generic, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "file" argument must be of type string. Received null
	at /home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771390
	at a (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771284)
	at w (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771035)
	at u (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:770559)
	at /home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771343
	at new Promise (<anonymous>)
	at l (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771322)
	at e.callInstallerScript (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:7215)
	at async H.value (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:763673)
	at async t.default.install (/home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/dist/extension.js:1:6773)
	at async /home/logos/.vscode/extensions/platformio.platformio-ide-2.3.3/dist/extension.js:1:12776
  ```

mraskin, 2021-09-21:
Hi,
I'm getting the exact same error for few weeks and did not manage to find the cause yet.
The good news is that I am no longer alone :-).

I'm on Linux Mint 18.3 Cinnamon.
Here are my messages

Error: TypeError [ERR_INVALID_ARG_TYPE]: The "file" argument must be of type string. Received null
	at /home/mraskin/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771390
	at a (/home/mraskin/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771284)
	at w (/home/mraskin/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:771035)
	at u (/home/mraskin/.vscode/extensions/platformio.platformio-ide-2.3.3/node_modules/platformio-node-helpers/dist/index.js:1:770559)
	at /home/mraskin/.vscode/extensions/platformio.platformio-ide-2.3.3/node...

Would be nice to find the cause, I'm currently stuck :-( .
Sorry to not bring a solution, but at least we got this issue on two different systems.

Thanks,
Michel.

JamesNewton, 2021-10-05:
Solution is known, but no one seems to follow the instructions: "BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN..."