platformio/platformio-core-installer #1472: Installation Manager

Kegadathechopper, 2021-10-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.2
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390
	at a (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
	at w (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035)
	at u (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559)
	at c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771343
	at new Promise (<anonymous>)
	at l (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771322)
	at e.callInstallerScript (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:7215)
	at async H.value (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:763673)
	at async t.default.install (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:6773)
	at async c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:12776
  ```

Kegadathechopper, 2021-10-05:
Why do i get this problem?

ivankravets, 2021-10-19:
Fixed in [PlatformIO IDE for VSCode v2.3.4](https://marketplace.visualstudio.com/items/platformio.platformio-ide/changelog). Please update PlatformIO IDE extension to the 2.3.4 version and restart VSCode.

Does it work for you now?