platformio/platformio-core-installer #1480: Installation Manager

Kegadathechopper, 2021-10-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.2
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390
	at a (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
	at w (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035)
	at u (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559)
	at c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771343
	at new Promise (<anonymous>)
	at l (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771322)
	at e.callInstallerScript (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:7215)
	at async H.value (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:763673)
	at async t.default.install (c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:6773)
	at async c:\Users\gavin\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:12776
  ```

Kegadathechopper, 2021-10-05:
why wont it lode Mane this prog is stupide

JamesNewton, 2021-10-05:
It probably doesn't load because you can't read: "BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN..."

Kegadathechopper, 2021-10-05:
and your a dick because you cant help!!  i did read and none of the solutions worked, SO why not FUCK OFF!! if you cant help.

Kegadathechopper, 2021-10-05:
Perhaps your just a dick in real life , who knows! however you think everyone has the same problem?or perhaps you just couldn't careless.
I did read, and no solution posed helped me. so i thought to post in case i had a slightly different problem!
did you STOP and think about that!!!  or is your email comment just so you can piss people off so you don't have to deal with anything?
maybe you should look for a different support option so yo can help and not be a dick.
Right!
as i said, i tried 5 different options and read helps suggestions none of which worked.
I did notice in my folder on the c: user"...  C:\Users\...\.platformio bl bla,, there was only one folder .cache nothing else?
i thought there would have been lot more..so clearly it cant read any files is what i thought..
what would your thought be Sir?
thanks

Kegadathechopper, 2021-10-06:
It probably doesn't work because you take so long to respond, and when you do you have nothing encouraging to say.and help other whom are looking for help.
SO GET A LIFE BEFORE ASUMING YOU KNOW EVERYTHING!!!