platformio/platformio-core-installer #1574: Installation Manager

IuliuNovac, 2021-10-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.60.2
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390
	at a (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
	at w (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035)
	at u (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559)
	at c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771343
	at new Promise (<anonymous>)
	at l (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771322)
	at e.callInstallerScript (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:7215)
	at async H.value (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:763673)
	at async t.default.install (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:6773)
	at async c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:12776
  ```

IuliuNovac, 2021-10-06:
I have python 4.10 installed already

BlueLightFR, 2021-10-06:
> I have python 4.10 installed already

uhm the last version of python is 3.10