platformio/platformio-core-installer #1704: Installation Manager

Sergi0Martin, 2021-10-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.59.1
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: TypeError [ERR_INVALID_ARG_TYPE]: The "path" argument must be of type string. Received null
	at c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771390
	at a (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771284)
	at w (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771035)
	at u (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:770559)
	at c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771343
	at new Promise (<anonymous>)
	at l (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:771322)
	at e.callInstallerScript (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:7215)
	at async H.value (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\index.js:1:763673)
	at async t.default.install (c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:6773)
	at async c:\Users\Sergio\.vscode\extensions\platformio.platformio-ide-2.3.3\dist\extension.js:1:12776
  ```

ivankravets, 2021-10-11:
Have you resolved this issue?

Sergi0Martin, 2021-10-13:
> Have you resolved this issue?

No, I had to install an older version.

ivankravets, 2021-10-13:
Older version of what?

ivankravets, 2021-10-13:
Could you contact me@ikravets.com dirctly? I'll need more details to reproduce this issue.