platformio/platformio-core-installer #1774: Installation Manager

ZengShifeng, 2021-10-21:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.61.2
  PIO IDE: v2.3.4
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Can not find Python Interpreter. Please install Python 3.6 or above manually
	at H.value (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:765899)
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
	at async H.value (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:764349)
	at async t.default.install (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\dist\extension.js:1:6773)
	at async c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\dist\extension.js:1:12776
  ```

ivankravets, 2021-10-21:
Could you run `platformio.exe` from `c:\Users\Administrator\.platformio\python3`? Does it work?

SaturdayScience, 2021-11-04:
Thank you Platformio seems to be installed OK

scorninpc, 2021-11-07:
I still can't install

[image]

[image]

I'm on Debian 11.1

I'v uninstall vscode, remove ~/.vscode and ~/platformio, reboot machine, but nothing works

scorninpc, 2021-11-07:
I found the problem. The recipe of AppImage has a problem, not VSCode. Installing de .deb all works

swanlm, 2022-01-09:
I am on macOS Version 12.0.1 (21A559).
The folder %HOME%/.platformio/python3/bin contains python3 but not python. Running python3 correctly starts up Python 3.8.9 (default, Jul 19 2021, 09:37:32) but since there is no python command I assume that 2.7 (the system default) is loaded. How do I correct this?

ivankravets, 2022-01-09:
@swanlm we don't use `%HOME%/.platformio/python3` directly. We use it to create Python virtual environment that is located in `%HOME%/.platformio/penv` and contains `python`.

Do you see any errors if try to install PlatformIO Core manually https://docs.platformio.org/en/latest/core/installation.html#super-quick-mac-linux ?

swanlm, 2022-01-09:
@ivankravets thanks for the quick response. I get no errors when installing PlatformIO Core manually. But as soon as I open vscode it wants to install PlatformIO IDE. At that point I get "PlatformIO: Can not find working Python 3.6+ Interpreter." Aborting installation gives me:
Error: Can not find Python Interpreter. Please install Python 3.6 or above manually at a.value (/Users/lanceswan/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:778120) at processTicksAndRejections (internal/process/task_queues.js:93:5) at async a.value (/Users/lanceswan/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:780890) at async S.install (/Users/lanceswan/.vscode/extensions/platformio.platformio-ide-2.4.0/dist/extension.js:1:4373) at async /Users/lanceswan/.vscode/extensions/platformio.platformio-ide-2.4.0/dist/extension.js:1:28136...

ivankravets, 2022-01-09:
Could you remove `%HOME%/.platformio/python3` and `%HOME%/.platformio/penv` folders and restart VSCode?

swanlm, 2022-01-09:
I get the same error. I am using vscode in conjunction with Fusion 360 and wonder if this related? I saw this in the console output.

console.ts:137 [Extension Host] /usr/bin/python3 Error: Python path configuration:
  PYTHONHOME = '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
  PYTHONPATH = '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.7/site-packages'
  program name = '/Library/Developer/CommandLineTools/usr/bin/python3'
  isolated = 0
  environment = 1
  user site = 1
  import site = 1
  sys._base_executable = '/Library/Developer/CommandLineTools/usr/bin/python3'
  sys.base_prefix = '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
  sys.base_exec_prefix = '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
  sys.executable = '/Library/Developer/CommandLineTools/usr/bin/python3'
  sys.prefix = '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
  sys.exec_prefix = '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
  sys.path = [
    '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.7/site-packages',
    '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python38.zip',
    '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.8',
    '/Users/lanceswan/Library/Application Support/Autodesk/webdeploy/production/88ce6bc1c34decd89efb26d037ca39bd2adac037/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.8/lib-dynload',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x0000000109ad2600 (most recent call first):
<no Python frame>

	at /Users/lanceswan/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:5240
	at ChildProcess.r (/Users/lanceswan/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)

ivankravets, 2022-01-09:
Sorry, we can't provide more help. It seems that Autodesk totally broke your system. They overwrote the system Python with something own broken. You can try to remove them from your PATH in init shell script.

swanlm, 2022-01-09:
Thanks; how would I do that, if you don't mind me asking?

ivankravets, 2022-01-09:
Please forward your question to https://community.platformio.org/ to keep this issue clean. Thanks for undersatding.

swanlm, 2022-01-09:
No problem and thanks for your help!

swanlm, 2022-01-09:
Hi, this resolved my problem: https://community.platformio.org/t/restricting-platformio-installation/12402

Specifically setting "Platformio-ide: Activate Only On Platform IOProject" to true.

ivankravets, 2022-01-09:
Ah... this is a custom Autodesk extension that breaks the whole `process.env` :(

apws, 2022-02-01:
hi, I just tried to run python get-platformio.py and it provided some error, I dont know why. But issue I have now is absolutelly F* CRAZY message "No pyvenv.cfg file" all the time yet after uninstalling and reainstalling python 310, into c:\ root or into user folder  (win10) ... I simply CANT RUN any f* python now .... please, can you help me?

ivankravets, 2022-02-01:
@apws 

> after connection broken by 'ProtocolError('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))'

Please fix this. Remove all Pythons. Restart VSCode.

apws, 2022-02-01:
thank you for reply, I somehow solved it for now by deleting and reinstalling .platformio through vscode - in fact, I am trying to use it rather with CLion (and I am new into pio, ya) while solving some build problems of clean arduino leonardo or zero projects - thank you again, will try more

p3moto, 2022-02-17:
> # Solution
> 
> If you use [PlatformIO IDE](https://platformio.org/platformio-ide), please ensure that [platformio-ide.useBuiltinPython](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-ide-usebuiltinpython) option is enabled (set to `true`).
> ## Windows and macOS
> 
> PlatformIO IDE comes with built-in portable Python 3 for Windows and macOS. **YOU DO NOT NEED** to install Python interpreter manually. If you use PlatformIO IDE and experience this issue, please navigate to `%HOME%/.platformio/python3` folder and check that `python.exe` (Windows) or `bin/python` works on your machine. In case if they don't work, please install manually the latest Python 3 using [this guide](https://docs.platformio.org/en/latest/faq.html#install-python-interpreter). Restart PlatformIO IDE (VScode).
> ## Linux
> 
> Please install the next package and restart IDE:
> 
> ```
> sudo apt-get install python3-venv
> ```
> 
> ## I still experience this issue
> 
> If you still experience this issue, please leave a comment below and describe your configuration (operating system, its version, etc.). We would be glad to help you.
> 
> Regards, Your friends at PlatformIO.
> 
> %23 Configuration
> 
> VSCode: 1.61.2 PIO IDE: v2.3.4 System: Windows_NT, 10.0.19043, x64
> 
> %23 Exception
> 
> ```
> Error: Can not find Python Interpreter. Please install Python 3.6 or above manually
>   at H.value (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:765899)
>   at runMicrotasks (<anonymous>)
>   at processTicksAndRejections (internal/process/task_queues.js:93:5)
>   at async H.value (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:764349)
>   at async t.default.install (c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\dist\extension.js:1:6773)
>   at async c:\Users\Administrator\.vscode\extensions\platformio.platformio-ide-2.3.4\dist\extension.js:1:12776
> ```

there is no posibillity to click on "home".... 
https://github.com/platformio/platformio-core-installer/issues/1967

dartrax, 2022-02-26:
Hi,
I have the same problem on Win 10 x64, what can I do?

> Could you run `platformio.exe` from `c:\Users\Administrator\.platformio\python3`? Does it work?

Already tried this, does not change anything.

Python works, the console opens when I click one of those files:
C:\Users\dartrax\.platformio\python3\python.exe
C:\Program Files\Python310\python.exe

It does not work when I click here (Windows Store opens):
C:\Users\dartrax\AppData\Local\Microsoft\WindowsApps\python.exe
C:\Users\dartrax\AppData\Local\Microsoft\WindowsApps\python3.exe

Environment variables are like this:
[image]

Thank you!

pazi88, 2022-02-28:
I have same issue as @dartrax Win10 x64, build 19042.1526. All did work just fine last week, but now it keeps complainen about Python and I can't seem to make it work, no matter what I do. Python works just fine outside PIO.

ivankravets, 2022-03-05:
@pazi88 could you remove `USERPROFILE_DIR/.platformio/penv` and `USERPROFILE_DIR/.platformio/python3` folders and restart VSCode? Do you use the latest versions of VSCode and PlatformIO IDE extension for VSCode?

dartrax, 2022-03-05:
@ivankravets, I can't speek for @pazi88, but in my case I do not have the penv directory (I think I already tried deleting that before) and deleting the python3 directory doesn't have any effect. I will be created again when I start VS Code again and the Error pops up.

ivankravets, 2022-03-06:
@dartrax what is your OS? Did you disable the built-in PlatformIO Core + Python interpreter for the PlatformIO IDE extension in the VSCode? Please check custom extension settings and enable built-in PIO Core + Python.

dartrax, 2022-03-06:
I have Win 10 x64, and built-in PlatformIO Core and Python are already enabled.

May be this screenshot helps, looks like if there is a syntax error somewhere...

[image]

ivankravets, 2022-03-06:
@dartrax Could you remove `%USERPROFILE%/.platformio/.cache` folder and restart VSCode?

dartrax, 2022-03-06:
@ivankravets Thank you for your help! Removing the .platformio/.cache folder finally helped!

pazi88, 2022-03-23:
> @dartrax Could you remove `%USERPROFILE%/.platformio/.cache` folder and restart VSCode?

Thanks! This helped for me too.

RFBomb, 2022-03-28:
[3095](https://github.com/platformio/platformio-vscode-ide/issues/3095) same issue. Cannot find interpreter, then subsequently fails. 

also %HOME% does nothing on windows 10.

dartrax, 2022-03-28:
For Windows, `%USERPROFILE%` instead of `%HOME%` should be used.

RFBomb, 2022-03-28:
> %USERPROFILE%\.platformio
is a completely empty folder

pukiva, 2022-04-25:
just got to c\user\ur username\.platformio
and delete the python3 and cache floder 
restard the vs code and re install the platformio

dartrax, 2022-04-25:
@pukiva, you surely mean c:\Users\ not c\user\.
However, @RFBomb already stated that this folder is already empty in his case.

donovan1983, 2022-06-21:
I had this issue on Windows 10 and tried the various troubleshooting steps with no luck. I previously had this version (2.4.3) of PlatformIO installed, so I knew there wasn't some system incompatibility.

I found that the file `python-portable-windows_amd64-1.30910.0.tar.gz` in `%USERPROFILE%/.platformio/.cache/tmp` was a 0-byte file, so of course the installer couldn't do anything with it.

After way too much searching, I found out the file can be found at [https://dl.registry.platformio.org/download/platformio/tool/python-portable/1.30910.0/python-portable-windows_amd64-1.30910.0.tar.gz]() and after downloading it and putting it in `%USERPROFILE%/.platformio/.cache/tmp`, PlatformIO was able to install.

ivankravets, 2022-06-22:
@donovan1983, thanks for sharing your experience. If you remove the `.cache` folder and restart VSCode, do you still see the 0-byte tar.gz archive?

Kamilake, 2022-07-11:
[image]
Same here.
I have a Python that works perfectly, but it doesn't work.

ivankravets, 2022-07-11:
If you have own global Python, please close VSCode and try Installer Script  https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

Do you see any errors?

Kamilake, 2022-07-13:
> If you have own global Python, please close VSCode and try Installer Script https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html
> 
> Do you see any errors?

The script seems to work without any problems. but
A warning message is still displayed that Python is not found.

[image]

Do you have any idea why?

Kamilake, 2022-07-13:
> The script seems to work without any problems. but A warning message is still displayed that Python is not found.
>
> [image]
>
> Do you have any idea why?

I answered myself. I solved it.
I had to run @ivankravets' script, remove the symbolic link of the Python env variable, reinstall the whole PIO, and empty that cache. 
I used Python from Windows store, found the path manually (`C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.1520.0_x64__qbz5n2kfra8p0\python.exe`) and added it to the environment variable and PYTHONHOME, and I succeeded in solving that without rebooting.
Thank you for your help!

Kamilake, 2022-07-13:
A new problem has occurred...  
 Each time you restart VSCode, the PIO downloads the portable Python and attempts to reinstall the PIO.

Kamilake, 2022-07-13:
[image]

If I turn on VSCode three times, it works well only once.
I decided to stop thinking.

donovan1983, 2022-07-18:
> @donovan1983, thanks for sharing your experience. If you remove the `.cache` folder and restart VSCode, do you still see the 0-byte tar.gz archive?

I had tried deleting the `.cache` folder multiple times, restarting VSCode multiple times, uninstalling and reinstalling PlatformIO, and even restarting my computer a couple of times and still ended up with the same results.

I forgot to note in my earlier comment that I have Anaconda Python installed in my global path and VSCode is set to use a virtual environment. Despite this, the PlatformIO installer claimed it couldn't find any Python interpreter to use and insisted on downloading its own portable Python.

Kamilake, 2022-08-16:
> [image]
> 
> When I turn on VSCode three times, it works well only ONCE. I decided to stop thinking. Throw it away!
> 
> (Currently, I'm checking the debug log and I will come back here if I can find out the cause or if there is any further update.)

still not working

JWSmythe, 2022-10-12:
I have a problem.  PlatformIO won't install.   It brought me to this issue to report it.

System:
OS: Windows 11 Pro 10.0.22622 Build 22622
Python installed with: python-3.10.8-amd64.exe
VSCode installed with: VSCodeSetup-x64-1.72.1.exe 
PlatformIO IDE extension attempted: v2.5.4

>>> Screen shots of the entire install process: https://imgur.com/a/xayEbuQ <<<
I only included one screenshot here, since you probably don't want 31 images wasting space here.   The complete chronological set is on imgur. 

--- BEGIN error text --- 
Error: Can not find Python Interpreter. Please install Python 3.6 or above manually at f.value (c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.5.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:801694) at process.processTicksAndRejections (node:internal/process/task_queues:96:5) at async f.value (c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.5.4-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:804843) at async E.install (c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.5.4-win32-x64\dist\extension.js:1:4485) at async c:\Users\User\.vscode\extensions\platformio.platformio-ide-2.5.4-win32-x64\dist\extension.js:1:27976...
--- END error text ---
[image]

It was working, but I hadn't used it in a month.  I loaded it up yesterday to do some work, and it didn't work.  
I tried lots of things to fix it. 

I've tried all the things.   

I've deleted all the things.  
I uninstalled VSCode and Python
I deleted my user .pio and .vscode directories.
I deleted some lingering directories that probably didn't hurt anything.

I just deleted and reinstalled everything, and took screen shots of absolutely every step involved, so you won't have to ask if I did a thing.

I tried not having any installed Python, so it would use the alleged internal portable version.  That didn't work. 
I tried installing Python before VSCode. 
I tried installing Python after VSCode. 
I tried installing both Python 3.9 and 3.10, in separate directories, and telling PIO specifically which to use.  That didn't work.

I installed the current Python with Python in %PATH%.   I downloaded it fresh, specifically for the 30th attempt, just in case.
I installed the current VSCode.  It was also downloaded fresh, just in case. 
I Trying to install the PlatformIO extension within VSCode, which fails.  
Because whatever test is happening to see if Python is there, isn't working. 

Previous attempts were made with the previous versions of Python and PlatformIO IDE extension.  They didn't help. 

All the steps, the full error text, and my relevant environment variables are all in the photos.   The important bits are above, so maybe it'll help someone else find it.  

Until then, I guess I'm using notepad++ this time.  ...again.  I'm trying really hard to want to like VSCode with PIO, but this is painful.  I've spent more time trying to make VSCode work, than I have actually coding with it.

ivankravets, 2022-10-12:
Do you use any espionage software? Kaspersky or another "antivirus" tool?

JWSmythe, 2022-10-12:
> 

Just Avast.  There's nothing in the Avast log indicating that anything was blocked.  It has done that in the past with other programs, but not with VSCode and Python.    If it had popped that warning up, I would have known what to fix.

JWSmythe, 2022-10-12:
Willing to be wrong, I disabled Avast and retried.  Same problem. 

[image]

ivankravets, 2022-10-13:
PlatformIO IDE comes with portable Python 3. Could you navigate to the `cd C:/Users/XXXXX/.platformio/python3` in `cmd.exe`, and type `python.exe`. Does it work for you?

JWSmythe, 2022-10-14:
It isn't getting that far.  The installer is aborting because it can't find python.  

[image]

ivankravets, 2022-10-15:
Do you see `.platformio` in the root of C drive?

JWSmythe, 2022-10-16:
That was it!   Thank you.  There was a 0 byte c:\.platformio file there.

I tried opening VSCode again, with the c:\.platformio file present.  It still errored.  So I closed it.
I renamed c:\.platformio , and re-ran VSCode.   It went straight to downloading the portable Python, and finished installing successfully. 

It's weird, that is an old file, Sept 11, 2022.   And it wasn't re-created during the new install.  Maybe whatever checks for the c:\.platformio file to be present, should check to see that whatever other required parts are still there.

[image]

ivankravets, 2022-10-16:
Was that a file or folder? (Broken .platformio)

JWSmythe, 2022-10-16:
It's a 0 byte file, not a directory.   Here's the info, looking at the file from a WSL.   It's still the same Windows machine.   I have only used the Windows installed PlatformIO, but I like looking at files and grepping from the Linux  WSL. :) 

I don't know what version of the PIO extension in VSCode it would have been though. I have uninstalled and reinstalled VSCode and PlatformIO several times since then, trying to resolve this problem.  It would have been fairly current, I like to keep software up to date.   If I got prompted to update, I would have updated. 

[image]

dzalf, 2022-10-19:
I am here to report a similar problem with pio and VSCode.

It had been working fine with an Arduino application until yesterday evening. I came this morning to my work PC and after installing a VSCode update pio seems to be broken.

I noticed that pio attempted reinstalling over and over even after removing the folders indicated in previous comments and reinstalling the extension.

After several attempts, the extension could be installed, however, I am seeing this error:

[image]

ZeoNish, 2023-02-12:
Open Vs Code Admin Mode! "run as administrator" !

ktmer02, 2023-06-12:
Im having problems with installing platformIO, im getting these error message.
I dont have the python3 exe in my .platformio folder is there a manual download?

[image]
[image]

JamesNewton, 2023-07-14:
I'm trying to get a bit more work out of an old Ubuntu 16.04 machine that I can't update right now. I was able to get everything working just fine on my Ubuntu 18 and 20 machines. But this 16.04 machine I can't seem to get platform io to see python3.6 (which is installed). 

VSCode 1.80.0

I do have the ~/.platformio/penv/bin folder, and a bunch of symlinks to python... but python3 is 3.5... I need to point to 3.6. So I tried the following, without luck. 

```
rm python
user@machine:~/.platformio/penv/bin$ ln -s /usr/bin/python3.6 python
user@machine:~/.platformio/penv/bin$ python --version
Python 2.7.12
user@machine:~/.platformio/penv/bin$ ./python --version
Python 3.6.13
user@machine:~/.platformio/penv/bin$ python3 --version
Python 3.5.2
user@machine:~/.platformio/penv/bin$ which python3
/usr/bin/python3
user@machine:~/.platformio/penv/bin$ ./python3 --version
Python 3.6.13
```
Is there a way to know which link it's following? I think it just always goes to python3 as in the global one, the `which python3` one, not what's linked in the ~/.platformio/penv/bin folder. 

So I tried this:
```
cd /usr/bin
user@machine:/usr/bin$ ls -l python*
lrwxrwxrwx 1 root root       9 Nov 23  2017 python -> python2.7
lrwxrwxrwx 1 root root       9 Nov 23  2017 python2 -> python2.7
-rwxr-xr-x 1 root root 3492624 Mar  1  2021 python2.7
lrwxrwxrwx 1 root root      33 Mar  1  2021 python2.7-config -> x86_64-linux-gnu-python2.7-config
lrwxrwxrwx 1 root root      16 Nov 23  2017 python2-config -> python2.7-config
lrwxrwxrwx 1 root root      25 Sep 13  2021 python3 -> /etc/alternatives/python3
-rwxr-xr-x 2 root root 4456208 Jan 26  2021 python3.5
-rwxr-xr-x 2 root root 4456208 Jan 26  2021 python3.5m
-rwxr-xr-x 2 root root 4723808 Feb 20  2021 python3.6
lrwxrwxrwx 1 root root      33 Feb 20  2021 python3.6-config -> x86_64-linux-gnu-python3.6-config
-rwxr-xr-x 2 root root 4723808 Feb 20  2021 python3.6m
lrwxrwxrwx 1 root root      34 Feb 20  2021 python3.6m-config -> x86_64-linux-gnu-python3.6m-config
-rwxr-xr-x 1 root root 5458744 Apr  8  2021 python3.9
lrwxrwxrwx 1 root root      33 Apr  8  2021 python3.9-config -> x86_64-linux-gnu-python3.9-config
lrwxrwxrwx 1 root root      10 Jul 28  2017 python3m -> python3.5m
lrwxrwxrwx 1 root root      16 Nov 23  2017 python-config -> python2.7-config
user@machine:/usr/bin$ sudo rm python3
user@machine:/usr/bin$ 
user@machine:/usr/bin$ sudo ln -s python3.9 python3
user@machine:/usr/bin$ python3 --version
Python 3.9.4
```
And now it at least gets past the "can't find python" thing... It's on to this error:
https://github.com/platformio/platformio-core-installer/issues/2851
which I will try to resolve tomorrow. Hope the machine still works with python3 set to 3.9

dw-liedji, 2023-07-15:
Installing ```sudo apt install -y python3-venv``` fixed the problem for me on ubuntu

ba58smith, 2023-07-21:
> Installing `sudo apt install -y python3-venv` fixed the problem for me on ubuntu

Me, too. Thanks!

rohanbari, 2023-07-30:
> Installing `sudo apt install -y python3-venv` fixed the problem for me on ubuntu

Dude we love you <3 thanks for saving my life

[27 more replies]