platformio/platformio-core-installer #1806: Installation Manager

andrub72, 2021-11-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.62.0
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\utente\.platformio\.cache\tmp\get-platformio-1.1.0.py", line 69, in <module>
    main()
  File "C:\Users\utente\.platformio\.cache\tmp\get-platformio-1.1.0.py", line 61, in main
    bootstrap()
  File "C:\Users\utente\.platformio\.cache\tmp\get-platformio-1.1.0.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\utente\.platformio\.cache\tmp\.piocore-installer-zf7q2xpw\tmp8thsok57\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "D:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\bin\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "D:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\bin\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "D:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\bin\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "D:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\bin\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 1260] Il programma � bloccato dai Criteri di gruppo. Per ulteriori informazioni, contattare l'amministratore del sistema
 -> Installer version: 1.1.0
Platform: Windows-10
Python version: 3.9.2 (tags/v3.9.2:1a79785, Feb 19 2021, 13:44:55) [MSC v.1928 64 bit (AMD64)]
Python path: D:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\bin\python3\python.exe
Creating a virtual environment at C:\Users\utente\.platformio\penv

	at d:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\data\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5240
	at ChildProcess.r (d:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\data\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (d:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable\data\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```
  IT'S POSSIBLE CHANGE DESTINATION FOR THIS FOLDER :
C:\Users\utente.platformio
in to
d:\AA_disco\PortableApps\PortableApps\Microsoft VS Code Portable.platformio

ivankravets, 2021-11-06:
> Error 1260: Windows cannot open this program because it has been prevented by a software restriction policy

Do you use 3rd party antivirus tools? I see a lot of hacks on your machine with portable software.

andrub72, 2021-11-06:
yes, there are limits on the user's folder, which is why I asked if it can
be set

ivankravets, 2021-11-06:
You can override PlatformIO's Core directory with this environment variable https://docs.platformio.org/en/latest/envvars.html#envvar-PLATFORMIO_CORE_DIR

andrub72, 2021-11-08:
in which folder is the platformio.ini file ??

ivankravets, 2021-11-08:
See https://docs.oracle.com/en/database/oracle/machine-learning/oml4r/1.5.1/oread/creating-and-modifying-environment-variables-on-windows.html#GUID-DD6F9982-60D5-48F6-8270-A27EC53807D0

Set `PLATFORMIO_CORE_DIR` to the folder where you have permissions.

closed 2021-11-08 by ivankravets

andrub72, 2021-11-08:
ok

andrub72, 2021-11-08:
I set the variable to this value ..
PLATFORMIO_CORE_DIR=D:\AA_disco\PortableApps\PortableApps\Microsoft VS Code
Portable\data\platformio

but now it gives me this error ?

"REPORT a problem" dont' work !!

 [image: image.png]