platformio/platformio-core-installer #1834: Installation Manager: Failed to install PlatformIO IDE

mjackdk, 2021-11-24:
%23 Description of problem
 Installation fails

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.62.3
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\jack\.platformio\.cache\tmp\get-platformio-1.1.0.py", line 69, in <module>
    main()
  File "C:\Users\jack\.platformio\.cache\tmp\get-platformio-1.1.0.py", line 61, in main
    bootstrap()
  File "C:\Users\jack\.platformio\.cache\tmp\get-platformio-1.1.0.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\pioinstaller\__main__.py", line 113, in core_check
  File "C:\Users\jack\.platformio\.cache\tmp\.piocore-installer-zmuygax5\tmp9vnv7mbk\pioinstaller.zip\pioinstaller\core.py", line 169, in check
  File "C:\Users\jack\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\jack\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\jack\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\jack\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
PermissionError: [WinError 5] Access is denied

	at c:\Users\jack\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5240
	at ChildProcess.r (c:\Users\jack\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\jack\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

mjackdk, 2021-11-25:
Closing this one, submitted #1836 with a cleaner environment

closed 2021-11-25 by mjackdk

zhang-and-M, 2022-01-14:
How to solve this problem

Muffinfreak2, 2022-01-17:
I have the same issue