platformio/platformio-core-installer #1958: PlatformIO: Can not find working Python 3.6+ Interpreter. Please install the latest Python 3 and restart VSCode

Jeebus678, 2022-02-13:
I've been using Platform:IO for well over a few years now, and suddenly one day it just stopped working. It's not detecting any python path, and even when I manually input the python path it just continues to give me the same prompt. I've read about this issue elsewhere and so far nothing. Currently using Python 3.10 installed from the official website, but I also tried installing locally through the Microsoft store. No luck. I have no idea why this suddenly stopped working.

Jeebus678, 2022-02-13:
I'm not entirely sure what fixed the issue but after a few hours of headache, it's functioning properly. What may have done it in the end:

- Disable built-in python interpreter
- Reload VScode
- Re-enable built-in python interpreterr
- Reload again

LucasChimento, 2022-06-13:
> No estoy del todo seguro de qué solucionó el problema, pero después de unas horas de dolor de cabeza, está funcionando correctamente. Lo que pudo haberlo hecho al final:
> 
> * Deshabilitar el intérprete de Python incorporado
> * Volver a cargar VScode
> * Vuelva a habilitar el intérprete de Python integrado
> * Volver a cargar

**Estoy con el mismo problema, ¿como es que deshabilitaste el interprete de Python integrado?**

Squizzy, 2022-10-15:
unblocking platformio stuff from comodo antivirus did it for me...

mehtivakili, 2023-08-23:
@LucasChimento for mine solved using this command and installing python-env because I had a python3.10 installed but it wasn't installed in the ubuntu 22.04 :
sudo apt-get install python3-venv
full thread: https://github.com/platformio/platformio-core-installer/issues/1774
but this issue mainly happens while installing the platformIO.

AncientTree, 2024-05-26:
> I'm not entirely sure what fixed the issue but after a few hours of headache, it's functioning properly. What may have done it in the end:
> 
> * Disable built-in python interpreter
> * Reload VScode
> * Re-enable built-in python interpreterr
> * Reload again

Hello @Jeebus678, would you tell me how to _Disable built-in python interpreter_? thanks