platformio/platformio-core-installer #1996: Compatible PlatformIO Core not found

Tom-Yucw, 2022-03-05:


ivankravets, 2022-03-05:
Do you use the latest versions of VSCode and PlatformIO IDE for VSCode?

Tom-Yucw, 2022-03-14:
Sorry for this answer comes so late.
I have found the reason.

In the plugin code of `await this.startInstaller(!hasPIOProject);`, main flow is described bellow:
`<- await this.startInstaller(!hasPIOProject);
<- const im = new InstallationManager(this.context.globalState, disableAutoUpdates);
<- new pioNodeHelpers.installer.PlatformIOCoreStage(......)
<- im.check()
<- await this.loadCoreState();
<- console.info(await callInstallerScript(await this.whereIsPython(), scriptArgs));   --here call get-platfrom.py
    // Load PIO Core state
    const coreState = await misc.loadJSON(stateJSONPath);
    console.info('PIO Core State', coreState);
    **core.setCoreState(coreState);  --here marked as pos1**

<- get-platfrom.py: pioinstaller.zip: core.py: _check_platform_version:
    state = penv.load_state()
    if state.get("platform") != platform.platform(terse=True):   --here, after some upgrade of ubuntu, platform.platform(terse=True) changes, which makes it not equal to the string stored in penv\state.json
        raise exception.InvalidPlatformIOCore(
            "PlatformIO Core was installed using another platform `%s`. "
            "Your current platform: %s"
            % (state.get("platform"), platform.platform(terse=True))
        )

---so, except is raised when calling get-platfrom.py, which cause **core.setCoreState(coreState);** in **pos1** is not executed, _CORE_STATE is not set`

When I toggled home page, 
` vscode.commands.registerCommand('platformio-ide.showHome', (startUrl) =>this.pioHome.toggle(startUrl))
 <- async toggle(startUrl = PIOHome.defaultStartUrl){
    this._currentPanel.webview.html = await this.getWebviewContent(startUrl);
  }
 <-  async getWebviewContent(startUrl){
    await pioNodeHelpers.home.ensureServerStarted
  }
 <- ensureServerStarted -> _ensureServerStarted -> runPIOCommand
 <- runPIOCommand{ await getCorePythonExe() }
 <- export async function getCorePythonExe() {
    const result = getCoreState().python_exe;   ---here, getCoreState will get **undefined|{}**, so no python_exe will be assigned
  }
 <-export async function getCorePythonExe() {
  const result = getCoreState().python_exe;
  if (!result) {
    throw new Error('Rivai Core is not installed');  --here raised the error for no python, but actually I had one
  }
  return result;
}`

ivankravets, 2022-06-22:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/644

closed 2022-06-22 by ivankravets