platformio/platformio-core-installer #2035: Installation Manager ... we need a solution for this problem, but nothing happen

Toddi3D, 2022-03-17:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.2
  PIO IDE: v2.4.2
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 69, in <module>
    main()
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 61, in main
    bootstrap()
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\Britti-Toddi\.platformio\.cache\tmp\.piocore-installer-m8_ea0fk\tmpmf0j8wfw\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Program Files\Inkscape\lib\python3.8\subprocess.py", line 1311, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Das System kann die angegebene Datei nicht finden
 -> Installer version: 1.1.1
Platform: Windows-10
Python version: 3.8.9 (default, Apr 13 2021, 15:54:59)  [GCC 10.2.0 64 bit (AMD64)]
Python path: C:\Program Files\Inkscape\bin\python.exe
Creating a virtual environment at C:\Users\Britti-Toddi\.platformio\penv

	at c:\Users\Britti-Toddi\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\index.js:1:5240
	at ChildProcess.c (c:\Users\Britti-Toddi\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\Britti-Toddi\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

Jay7286, 2022-03-19:
There is no duplicates found in above link

closed 2022-04-27 by stale[bot]

juapillo, 2022-06-19:
%23 Description of problem
  Problem with fresh installation of the plugin on VS Core System installation (on c:\Program Files)

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.4.3
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\juanj\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 69, in <module>
    main()
  File "C:\Users\juanj\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 61, in main
    bootstrap()
  File "C:\Users\juanj\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\juanj\.platformio\.cache\tmp\.piocore-installer-v_oibpte\tmpffhq_vbj\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Users\juanj\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\juanj\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\juanj\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\juanj\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
 -> Installer version: 1.1.1
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\juanj\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\juanj\.platformio\penv

	at c:\Users\juanj\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:5240
	at ChildProcess.c (c:\Users\juanj\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\juanj\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```