platformio/platformio-core-installer #2149: Can't Install even with python3-venv

Chandler-Kluser, 2022-05-18:
## Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 
   
  -> [Screenshot of PlatformIO Error Message](https://imgur.com/a/m0DG5kj)
## Configuration

  VSCode: 1.67.1
  PIO IDE: v2.4.3
  System: Linux, 5.15.0-30-generic, x64
  OS: Ubuntu 22.04 LTS
  
   -> [Screenshot of my system neofetch](https://i.imgur.com/Jq23TSd.png)

  ## Exception
  ```
  Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.1.1
Platform: Linux-5.15.0-30-generic-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Apr  2 2022, 09:04:19) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/chandler/.platformio/penv

	at /home/chandler/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:5240
	at ChildProcess.c (/home/chandler/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

## Notes

I DO have python3-venv installed in my two python interpreters (3.10 python3 AND 3.9 - python), checked with:
```bash
# python3
sudo apt install python3-venv
pip3 install virtualenv
# python
pip install virtualenv
```
I have also manually created a virtual environment at where platformio asked for
```bash
python3 -m venv /home/chandler/.platformio/penv
```
and then reload VS Code to install Platform IO Core again, but the issue persists.
I have also tried unchecking the option `Use Builtin Python` and then reloaded VS Code to try again to install PlatformIO Core.

-> [Screenshot with the console in VS Code Developer Tools Toggled](https://imgur.com/a/WMsvICd)

SaeedRezaAmanat, 2022-05-28:
Have you used sudo apt-get install python3-venv ?
#85 
https://issuemode.com/issues/platformio/platformio-core-installer/42084394

Chandler-Kluser, 2022-06-19:
yes, I have also tried to use it with a brand new virtual environment but also without success

ivankravets, 2022-06-29:
Could you reproduce this issue again?

Chandler-Kluser, 2022-07-10:
Yes, I can. First Step is going to extensions and install PlatformIO IDE:
[image]
Second Step is wait for the installation of the extension, which I get this error message:
[image]
I have already python3-venv installed:
[image]
My python3 is in PATH and it is in the `/usr/bin` folder:
[image]
I already have a pyenv in home folder with these binaries:
[image]
I have already delete the .platformio folder + reinstalling the extension, nothing worked.

ivankravets, 2022-07-11:
Could you close VSCoe and try installer script https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html

Do you see any errors?

Chandler-Kluser, 2022-07-12:
[image]

Chandler-Kluser, 2022-08-11:
closing, but the issue is not solved

ivankravets, 2022-08-12:
I've just installed a clean latest Ubuntu in my VM and can't reproduce this issue. Could you provide an output from the `python3 get-platformio.py`. The GIF which you provided closes TOO FAST, and I can't read the error at the end.

Chandler-Kluser, 2022-08-21:
I have done a [short video](https://drive.google.com/file/d/19OL2jsam9yiPZ3WD88dbI-Zc4nUhlT68/view?usp=sharing)

closed 2022-09-20 by Chandler-Kluser

freynder, 2023-07-20:
same issue here using debian

James-Livesey, 2023-07-23:
Removing Python versions above 3.8 (as what @freynder suggested) works for me!