platformio/platformio-core-installer #2214: Installation Manager

DROBNJAK, 2022-06-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.2
  PIO IDE: v2.5.0
  System: Windows_NT, 6.1.7601, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Windows-7
Python version: 3.7.7 (tags/v3.7.7:d7c567b08f, Mar 10 2020, 10:41:24) [MSC v.1900 64 bit (AMD64)]
Python path: C:\Users\Dejan Corovic\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Dejan Corovic\.platformio\penv

	at c:\Users\Dejan Corovic\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5227
	at ChildProcess.c (c:\Users\Dejan Corovic\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\Dejan Corovic\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-29:
Could you reproduce this issue again?

DROBNJAK, 2022-06-29:
Thanks for coming back.

It seems that it is an intermittent problem, that depends on the internet connection speed / CPU overload. In my second attempt it worked OK.

As well, it seemed that anti-virus software was registering Python.exe as malware?

But it all works now.

ivankravets, 2022-06-29:
Please uninstall russian’s espionage antivirus software. It brings more evil to your machine than protects you.

closed 2022-06-29 by ivankravets