platformio/platformio-core-installer #2219: Installation Manager

rwtools, 2022-06-25:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Linux, 4.19.0-20-amd64, x64

  %23 Exception
  ```
  Error: Error: Can not install PlatformIO Core due to a missed `distutils` package in your Python installation.
Please install this package manually using the OS package manager. For example:

$ apt-get install python3-venv

(MAY require administrator access `sudo`)
 -> Installer version: 1.1.2
Platform: Linux-4.19.0-20-amd64-x86_64-with-glibc2.28
Python version: 3.9.1 (default, May  1 2022, 18:32:15) 
[GCC 8.3.0]
Python path: /usr/bin/python
Creating a virtual environment at /home/rainer/.platformio/penv

	at /home/rainer/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/home/rainer/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

Rabbible, 2022-06-25:
After changing another Internet connection and downloading the newest python (python3.10.5),problem is soloved.

rwtools, 2022-06-25:
Thank you,
Solved it by upgrading to Debian Bullseye.

closed 2022-06-29 by ivankravets