platformio/platformio-core-installer #2223: Installation Manager

mgagvani, 2022-06-25:
**Description of problem**

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

**Configuration**

  VSCode: 1.67.2
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  **Exception**
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\username\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\username\.platformio\penv
C:\Users\username\.platformio\python3\python.exe

	at c:\Users\username\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5227
	at ChildProcess.c (c:\Users\username\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\username\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-29:
Could you reproduce this issue again?

mgagvani, 2022-07-03:
No I can't. It might have had to do with VSCode updating and restarting but I do not have any more information.

closed 2022-07-03 by mgagvani