platformio/platformio-core-installer #2234: Installation Manager

ask6483, 2022-06-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Linux, 5.18.0-2-amd64, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/ask/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/ask/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/ask/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/ask/.platformio/.cache/tmp/.piocore-installer-l2ircklh/tmptz9p79rq/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/ask/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: Linux-5.18.0-2-amd64-x86_64-with-glibc2.33
Python version: 3.10.5 (main, Jun  8 2022, 09:26:22) [GCC 11.3.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/ask/.platformio/penv

	at /home/ask/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/home/ask/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-29:
Could you reproduce this issue again?

ask6483, 2022-07-21:
Yes over na over, tried uninstalling no luck.

ask6483, 2022-07-21:
Deleted ~/.platformio directory not working either.

zingo, 2022-09-18:
Seem to be the same as https://github.com/platformio/platformio-core-installer/issues/2404

zingo, 2022-09-20:
Installing python3-venv as mentioned in https://github.com/platformio/platformio-core-installer/issues/85
solved it for me on popOS! based on Ubuntu 22.04.

closed 2022-11-01 by stale[bot]