platformio/platformio-core-installer #2239: Installation Manager

antoinnneee, 2022-06-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.22581, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\antoi\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\antoi\.platformio\penv

    at cmd (c:\Users\antoi\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:187:25)
    at ChildProcess.c (c:\Users\antoi\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (node:events:390:28)
    at ChildProcess.emit (node:domain:475:12)
    at ChildProcess.cp.emit (c:\Users\antoi\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (node:internal/child_process:1064:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-29:
Could you reproduce this issue again?

antoinnneee, 2022-06-29:
ok, that was due to weird problem, I remove plateformio and reinstall, same error, I made the same things with Visual Studio and same issue.
Finaly I change the python version from the latest (3.10.5) to 3.9.13 versions and all the issue are solved

closed 2022-06-29 by antoinnneee