platformio/platformio-core-installer #2240: Installation Manager

Jared-DEDW, 2022-06-29:
%23 Description of problem
All was working well, then I created a VSCode workspace and switched to it. After that I started getting this error. Now I get this error even when not using the new workspace. PlatformIO seems to just not open. 

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.0-insider
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\Me\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\Me\.platformio\penv

	at c:\Users\Me\.vscode-insiders\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5227
	at ChildProcess.c (c:\Users\Me\.vscode-insiders\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\Me\.vscode-insiders\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-29:
Could you reproduce this issue again?

Jared-DEDW, 2022-06-29:
I am in the middle of an active project and do not want to risk messing it up again at this time by attempting to recreate the issue.

ivankravets, 2022-06-29:
When you will have spare time, I would be thankful if you try to reproduce this issue. You can remove the `.platformio` folder at all and PlatformIO will create a new and download all related packages.