platformio/platformio-core-installer #2246: Installation Manager

DikiFive, 2022-06-30:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.
	at u.value (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:801864)
	at S.install (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:4373)
	at c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27870
	at E._withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:81:44253)
	at E.withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:81:44007)
	at Object.withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:97:42210)
	at Object.startInstaller (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27285)
	at async Object.activate (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:22733)
  ```

ivankravets, 2022-06-30:
Could you reproduce this issue again?

DikiFive, 2022-07-01:
Did you mean you need I replace this problem again?Or just Submit the other issuse

DikiFive, 2022-07-01:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.
	at u.value (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:801864)
	at S.install (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:4373)
	at c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27870
	at E._withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:81:44253)
	at E.withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:81:44007)
	at Object.withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:97:42210)
	at Object.startInstaller (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27285)
	at processTicksAndRejections (node:internal/process/task_queues:96:5)
	at async Object.activate (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:22733)
  ```

ivankravets, 2022-07-01:
Could you provide a remote session to debug this issue? Please contact me at me@ikravets.com.

DikiFive, 2022-07-02:
Sorry,I'm in China, you know, maybe Internet connection may be a problem. I'd better ask for help on the Chinese Forum, thank you.

ivankravets, 2022-07-02:
Could you try the next links and tell us the download speed?

1. https://dl.registry.nm1.platformio.org/tools/11/a2/760f0689ed9540b8683b6971b2a20ec46a9c801fd3cc7fdeba5de6726258/python-portable-windows_amd64-1.30910.0.tar.gz
2. https://eu2.contabostorage.com/28afe64ba7f44add9b788964b0b1907a:pioregmirror/tools/11/a2/760f0689ed9540b8683b6971b2a20ec46a9c801fd3cc7fdeba5de6726258/python-portable-windows_amd64-1.30910.0.tar.gz
3. https://github.com/ivankravets/pioreg-ci-mirror/releases/download/python-portable-windows_amd64/11a2760-python-portable-windows_amd64-1.30910.0.tar.gz

DikiFive, 2022-07-02:
250kb/s

ivankravets, 2022-07-02:
For 1, 2 or 3?

DikiFive, 2022-07-02:
1 and 3 ,2 only 100kb/s

ivankravets, 2022-07-02:
@Hjj-wing, could you share your results? This information will help us to tune our CDN.

https://github.com/platformio/platformio-vscode-ide/issues/3234

ivankravets, 2022-07-02:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2246#issuecomment-1172891319

DikiFive, 2022-07-02:
So,how can I save this question. I've been troubled by this for a long time。Could you give some advice？Thank you very much

ivankravets, 2022-07-02:
We have just made some improvements. Could you remove `c:\Users\14185\.platformio\.cache` folder and restart VSCode? Does it work now?

DikiFive, 2022-07-02:
let me try.wait a minute

DikiFive, 2022-07-02:
It seems that it can't be solved. This is the latest error report

DikiFive, 2022-07-02:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.
	at u.value (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:801864)
	at S.install (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:4373)
	at c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27870
	at E._withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:81:44253)
	at E.withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:81:44007)
	at Object.withProgress (c:\Users\14185\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:97:42210)
	at Object.startInstaller (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27285)
	at async Object.activate (c:\Users\14185\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:22733)
  ```

ivankravets, 2022-07-02:
How can we help you here if you disabled PlatformIO Core? Please open VSCode settings and enable useBuiltinPIOCore

closed 2022-07-02 by ivankravets

DikiFive, 2022-07-02:
I tried to open the Core, but I got stuck in the download and installation process

DikiFive, 2022-07-02:
The installation progress bar did not move, so I tried to disable it on the Chinese Forum, but it still failed

DikiFive, 2022-07-02:
[image]

DikiFive, 2022-07-02:
This Core can  Offline installation？

DikiFive, 2022-07-02:
[image]
Now it seems that the installation has been successful, but the homepage has not been opened

ivankravets, 2022-07-02:
Do you have antivirus in the system?

DikiFive, 2022-07-02:
No, none. I even shut down Microsoft defender

ivankravets, 2022-07-02:
Could you click on the terminal icon on the bottom status  bar and type:
```
pio home
```

Does it work? The PIO Home should be opened in your default browser.

DikiFive, 2022-07-03:
[image]
Look at this.I think that maybe failed to install successfully.

ivankravets, 2022-07-03:
You didn’t open PIO Core CLI. See https://docs.platformio.org/integration/ide/vscode.html