platformio/platformio-core-installer #2250: Installation Manager

minihant, 2022-07-01:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\user\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 68, in <module>
    main()
  File "C:\Users\user\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 60, in main
    bootstrap()
  File "C:\Users\user\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 46, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\Users\user\.platformio\.cache\tmp\.piocore-installer-k7a20imh\tmp6by7lb0b\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Users\user\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\user\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\user\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\user\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] �t�Χ䤣����w���ɮסC
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\user\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\user\.platformio\penv

	at c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5227
	at ChildProcess.c (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\user\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-01:
Could you reproduce this issue after restarting VSCode?

zingo, 2022-09-18:
Seem to be the same as https://github.com/platformio/platformio-core-installer/issues/2404

closed 2022-11-01 by stale[bot]