platformio/platformio-core-installer #2259: Installation Manager

ChasterKras, 2022-07-04:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Linux, 5.13.0-52-generic, x64

  %23 Exception
  ```
  Error: Error: Compatible PlatformIO Core not found.
Reason: Could not run `/home/bohdan/.platformio/penv/bin/platformio --help`.
Error: Error: Traceback (most recent call last):
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1054, in main
    with self.make_context(prog_name, args, **extra) as ctx:
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 920, in make_context
    self.parse_args(ctx, args)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1613, in parse_args
    rest = super().parse_args(ctx, args)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1378, in parse_args
    value, args = param.handle_parse_result(ctx, opts, args)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 2360, in handle_parse_result
    value = self.process_value(ctx, value)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 2322, in process_value
    value = self.callback(ctx, self, value)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1273, in show_help
    echo(ctx.get_help(), color=ctx.color)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 699, in get_help
    return self.command.get_help(self)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1298, in get_help
    self.format_help(ctx, formatter)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1331, in format_help
    self.format_options(ctx, formatter)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1533, in format_options
    self.format_commands(ctx, formatter)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1586, in format_commands
    cmd = self.get_command(ctx, subcommand)
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/cli.py", line 80, in get_command
    module = importlib.import_module(commands[cmd_name])
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/commands/ci.py", line 24, in <module>
    from platformio.project.commands.init import project_init_cmd, validate_boards
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/project/commands/init.py", line 55, in <module>
    @click.option("--ide", type=click.Choice(ProjectGenerator.get_supported_ides()))
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/project/integration/generator.py", line 56, in get_supported_ides
    [
  File "/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/project/integration/generator.py", line 56, in <listcomp>
    [
  File "/usr/lib/python3.8/pathlib.py", line 1122, in iterdir
    for name in self._accessor.listdir(self):
FileNotFoundError: [Errno 2] No such file or directory: '/home/bohdan/.platformio/penv/lib/python3.8/site-packages/platformio/project/integration/tpls'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/bohdan/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/home/bohdan/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Socket.<anonymous> (node:internal/child_process:450:11)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2022-07-04:
Thanks for the report, fixed! Please remove `~/.platformio/penv` folder and restart VSCode.

closed 2022-07-04 by ivankravets

ChasterKras, 2022-07-04:
https://github.com/platformio/platformio-vscode-ide/issues/205#issuecomment-893571757

ChasterKras, 2022-07-04:
@ivankravets , yes. Install 3.7.11 python env helped me fix this problem. Thank you, for quickly answer

ivankravets, 2022-07-04:
What is your default Python? What are your OS and its version?

ChasterKras, 2022-07-04:
> What is your default Python? What are your OS and its version?

default Python was 3.9, Ubuntu 20.04

marcovannoord, 2022-07-04:
Removing the /penv folder and restarting it resolved the issue

ivankravets, 2022-07-04:
@ChasterKras which issues did you have with Python 3.9? Could re-test again with https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html ?