platformio/platformio-core-installer #2260: Installation Manager

SujithChristopher, 2022-07-04:
was trying to install in vs code, this error popped out

%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\CMC\.platformio\python3\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\CMC\.platformio\python3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\CMC\.platformio\penv\lib\site-packages\pip\__main__.py", line 31, in <module>
    sys.exit(_main())
  File "C:\Users\CMC\.platformio\penv\lib\site-packages\pip\_internal\cli\main.py", line 68, in main
    command = create_command(cmd_name, isolated=("--isolated" in cmd_args))
  File "C:\Users\CMC\.platformio\penv\lib\site-packages\pip\_internal\commands\__init__.py", line 109, in create_command
    module = importlib.import_module(module_path)
  File "C:\Users\CMC\.platformio\python3\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\CMC\.platformio\penv\lib\site-packages\pip\_internal\commands\install.py", line 14, in <module>
    from pip._internal.cli.req_command import (
  File "C:\Users\CMC\.platformio\penv\lib\site-packages\pip\_internal\cli\req_command.py", line 20, in <module>
    from pip._internal.index.collector import LinkCollector
  File "C:\Users\CMC\.platformio\penv\lib\site-packages\pip\_internal\index\collector.py", line 14, in <module>
    import xml.etree.ElementTree
ModuleNotFoundError: No module named 'xml.etree.ElementTree'
Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\CMC\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\CMC\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\CMC\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core

    at cmd (c:\Users\CMC\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:187:25)
    at ChildProcess.c (c:\Users\CMC\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (node:events:390:28)
    at ChildProcess.emit (node:domain:475:12)
    at ChildProcess.cp.emit (c:\Users\CMC\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (node:internal/child_process:1064:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-04:
Please remove `C:\Users\CMC\.platformio` folder and restart VSCode. Does it work now?

SujithChristopher, 2022-07-04:
yes

closed 2022-07-04 by SujithChristopher