platformio/platformio-core-installer #2265: Installation Manager

PippoAe, 2022-07-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
 Error: Error: Compatible PlatformIO Core not found. Reason: Could not run `C:\Users\phili\.platformio\penv\Scripts\platformio.exe --help`. Error: Error: Traceback (most recent call last): File "C:\Users\phili\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 102, in main cli() # pylint: disable=no-value-for-parameter ^^^^^ File "C:\Users\phili\.platformio\penv\Lib\site-packages\click\core.py", line 1130, in __call__ return self.main(*args, **kwargs) ^^^^^^^^^^^^^^^^^^^^^^^^^^ File "C:\Users\phili\.platformio\penv\Lib\site-packages\click\core.py", line 1054, in main with self.make_context(prog_name, args, **extra) as ctx: ...
 

Happends during "Installation PlatformIO Core" step in VS Code.
Uninstalling and reinstalling platform.IO did not help....

I might have changed some python system-paths to build another project... but other than that, no clue what could cause the problem.

Any help would be greatly appreciated.

![2022-07-06_22h35_56](https://user-images.githubusercontent.com/20857236/177638495-f343b598-a7a9-40cf-845c-408bb1ba0f66.png)

PippoAe, 2022-07-06:
deleting "penv"-Folder did not change.

"python" in cmd returns the following:
[image]

ivankravets, 2022-07-07:
We didn't test PlatformIO with the beta Python 3.11. Please use stable Python and uninstall Python 3.11-dev.

PippoAe, 2022-07-08:
Worked, thank you! <3