platformio/platformio-core-installer #2272: Installation Manager

StevenArkon, 2022-07-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.0
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "C:\Users\Steven\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "C:\Users\Steven\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "C:\Users\Steven\.platformio\.cache\tmp\get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\core.py", line 1659, in invoke
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\pioinstaller\__main__.py", line 113, in core_check
  File "C:\Users\Steven\.platformio\.cache\tmp\.piocore-installer-eqbx465l\tmpld3ud3ez\pioinstaller.zip\pioinstaller\core.py", line 169, in check
  File "C:\Users\Steven\.platformio\python3\lib\subprocess.py", line 424, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Steven\.platformio\python3\lib\subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\Steven\.platformio\python3\lib\subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\Steven\.platformio\python3\lib\subprocess.py", line 1420, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
PermissionError: [WinError 5] Access is denied

	at c:\Users\Steven\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5227
	at ChildProcess.c (c:\Users\Steven\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\Steven\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

StevenArkon, 2022-07-07:
uninstalled and reinstalled

ivankravets, 2022-07-08:
Does it work now?

closed 2022-09-25 by stale[bot]