platformio/platformio-core-installer #2273: Installation Manager. FileNotFoundError .platformio/penv/bin/python

Malangcow, 2022-07-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.70.0-insider
  PIO IDE: v2.5.0
  System: Linux, 5.17.15-76051715-generic, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/seungwon/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/seungwon/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/seungwon/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/seungwon/.platformio/.cache/tmp/.piocore-installer-vomt7gbb/tmpdoy8v6ag/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/seungwon/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: Linux-5.17.15-76051715-generic-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Apr  2 2022, 09:04:19) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/seungwon/.platformio/penv

    at cmd (/home/seungwon/.vscode-insiders/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:187:25)
    at ChildProcess.c (/home/seungwon/.vscode-insiders/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:177:7)
    at ChildProcess.emit (node:events:390:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1064:16)
    at Socket.<anonymous> (node:internal/child_process:450:11)
    at Socket.emit (node:events:390:28)
    at Socket.emit (node:domain:475:12)
    at Pipe.<anonymous> (node:net:687:12)
  ```

This is tree output of my .platformio directory.

```
❯ exa -dTL 4 $PWD -l
drwxrwxr-x    - seungwon  9 Jul 18:32 /home/seungwon/.platformio
drwxrwxr-x    - seungwon  9 Jul 18:32 └── penv
drwxrwxr-x    - seungwon  9 Jul 18:32    ├── lib
drwxrwxr-x    - seungwon  9 Jul 18:32    │  └── python3.10
drwxrwxr-x    - seungwon  9 Jul 18:32    ├── local
drwxrwxr-x    - seungwon  9 Jul 18:32    │  ├── bin
.rw-rw-r-- 2.2k seungwon  9 Jul 18:32    │  │  ├── activate
.rw-rw-r-- 1.4k seungwon  9 Jul 18:32    │  │  ├── activate.csh
.rw-rw-r-- 3.0k seungwon  9 Jul 18:32    │  │  ├── activate.fish
.rw-rw-r-- 2.6k seungwon  9 Jul 18:32    │  │  ├── activate.nu
.rw-rw-r-- 1.8k seungwon  9 Jul 18:32    │  │  ├── activate.ps1
.rw-rw-r-- 1.2k seungwon  9 Jul 18:32    │  │  ├── activate_this.py
.rw-rw-r--  682 seungwon  9 Jul 18:32    │  │  ├── deactivate.nu
.rwxrwxr-x  253 seungwon  9 Jul 18:32    │  │  ├── pip
.rwxrwxr-x  253 seungwon  9 Jul 18:32    │  │  ├── pip-3.10
.rwxrwxr-x  253 seungwon  9 Jul 18:32    │  │  ├── pip3
.rwxrwxr-x  253 seungwon  9 Jul 18:32    │  │  ├── pip3.10
lrwxrwxrwx   16 seungwon  9 Jul 18:32    │  │  ├── python -> /usr/bin/python3
lrwxrwxrwx    6 seungwon  9 Jul 18:32    │  │  ├── python3 -> python
lrwxrwxrwx    6 seungwon  9 Jul 18:32    │  │  ├── python3.10 -> python
.rwxrwxr-x  240 seungwon  9 Jul 18:32    │  │  ├── wheel
.rwxrwxr-x  240 seungwon  9 Jul 18:32    │  │  ├── wheel-3.10
.rwxrwxr-x  240 seungwon  9 Jul 18:32    │  │  ├── wheel3
.rwxrwxr-x  240 seungwon  9 Jul 18:32    │  │  └── wheel3.10
drwxrwxr-x    - seungwon  9 Jul 18:32    │  └── lib
drwxrwxr-x    - seungwon  9 Jul 18:32    │     └── python3.10
.rw-rw-r--  203 seungwon  9 Jul 18:32    └── pyvenv.cfg
```

As you can see `/home/seungwon/.platformio/penv/bin/python` does not exist.

What I've tried ( None of them worked )
1. Restarting of vscode after PlatformIO IDE install fail
2. Uninstall and reinstall PlatformIO IDE vscode extension after the install fail
3. Uninstall PlatformIO IDE vscode extension, `rm -fr .platformio`  reinstall PlatformIO IDE vscode extension

ivankravets, 2022-07-09:
Could you try the installer script and share an output here? [Local Download (macOS / Linux / Windows)](https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#id2)

Malangcow, 2022-07-09:
@ivankravets Sure. Here's the output.

```
❯ wget https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
--2022-07-10 08:04:27--  https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 774690 (757K) [text/plain]
Saving to: ‘get-platformio.py’

get-platformio.py   100%[===================>] 756.53K  3.95MB/s    in 0.2s    

2022-07-10 08:04:29 (3.95 MB/s) - ‘get-platformio.py’ saved [774690/774690]

❯ python3 get-platformio.py
Installer version: 1.1.2
Platform: Linux-5.17.15-76051715-generic-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Apr  2 2022, 09:04:19) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/seungwon/.platformio/penv
Traceback (most recent call last):
  File "/home/seungwon/Downloads/get-platformio.py", line 68, in <module>
    main()
  File "/home/seungwon/Downloads/get-platformio.py", line 60, in main
    bootstrap()
  File "/home/seungwon/Downloads/get-platformio.py", line 46, in bootstrap
    pioinstaller.__main__.main()
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/seungwon/Downloads/.piocore-installer-lo1jbgwc/tmpanr3i0ag/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/seungwon/.platformio/penv/bin/python'
```

Making alias for python3 didn't work either.
( Adding `alias python=python3` in the `.zshrc` )

zingo, 2022-09-18:
> Please remove /home/seungwon/.platformio/penv folder and restart VSCode. Does it work now?
Still the same problem

Could be the same problem as https://github.com/platformio/platformio-core-installer/issues/2404

zingo, 2022-09-20:
Installing python3-venv as mentioned in https://github.com/platformio/platformio-core-installer/issues/85
solved it for me on popOS! based on Ubuntu 22.04.

closed 2022-11-01 by stale[bot]