platformio/platformio-core-installer #2275: Installation Manager

ademenev, 2022-07-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.0
  PIO IDE: v2.5.0
  System: Darwin, 19.6.0, x64

  %23 Exception
  ```
  Error: /Users/andrey/.platformio/penv/bin/python: No module named pip
Error: Could not install PlatformIO Core: Command '['/Users/andrey/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: macOS-10.15.7
Python version: 3.9.9 (main, Nov 29 2021, 12:55:48) 
[Clang 10.0.0 (clang-1000.10.44.4)]
Python path: /Users/andrey/.platformio/python3/bin/python3
Creating a virtual environment at /Users/andrey/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core

	at /Users/andrey/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/Users/andrey/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Socket.<anonymous> (node:internal/child_process:450:11)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2022-07-11:
Please remove `/Users/andrey/.platformio/penv` folder and restart VSCode. Does it work now?

ademenev, 2022-07-11:
> Please remove `/Users/andrey/.platformio/penv` folder and restart VSCode. Does it work now?

Nope. After removing penv and restarting, a notification is displayed about installing portable Python, then it changes to "installing PlatformIO core", and after a while the same error is reported

ivankravets, 2022-07-11:
Could you try installer script https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html and provide errors here?

ademenev, 2022-07-11:
```
andrey@Mac-mini ~ % python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/master/scripts/get-platformio.py)"
Installer version: 1.1.2
Platform: macOS-10.15.7
Python version: 3.9.13 (main, May 24 2022, 21:28:12)
[Clang 12.0.0 (clang-1200.0.32.29)]
Python path: /usr/local/opt/python@3.9/bin/python3.9
Creating a virtual environment at /Users/andrey/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
/Users/andrey/.platformio/penv/bin/python: No module named pip
Error: Could not install PlatformIO Core: Command '['/Users/andrey/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
Traceback (most recent call last):
  File "<string>", line 105, in <module>
  File "<string>", line 101, in main
  File "/usr/local/Cellar/python@3.9/3.9.13_1/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/local/opt/python@3.9/bin/python3.9', '/var/folders/vq/2m3w1y_j0p3fwr2yv84l62740000gn/T/tmp2ouq6g2n']' returned non-zero exit status 1.
andrey@Mac-mini ~ %
```

ivankravets, 2022-07-15:
Please provide an output of the following command:

```
ls -al /Users/andrey/.platformio/penv/bin
```

closed 2022-09-27 by stale[bot]