platformio/platformio-core-installer #2276: Installation Manager

MarkusSmartHome, 2022-07-09:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Darwin, 18.7.0, x64

  %23 Exception
  ```
  Error: Could not install packages due to an EnvironmentError: Could not find a suitable TLS CA certificate bundle, invalid path: /Users/markusgschwend/.platformio/penv/lib/python3.7/site-packages/pip/_vendor/certifi/cacert.pem

Error: Could not install PlatformIO Core: Command '['/Users/markusgschwend/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Darwin-18.7.0
Python version: 3.7.3 (v3.7.3:ef4ec6ed12, Mar 25 2019, 16:52:21) 
[Clang 6.0 (clang-600.0.57)]
Python path: /Library/Frameworks/Python.framework/Versions/3.7/bin/python3
Creating a virtual environment at /Users/markusgschwend/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Downloading https://files.pythonhosted.org/packages/71/d7/6793c4c903d373df2223bb9e89bb943b415bbd8e249ff47fe1500cfea513/platformio-6.1.0.tar.gz (246kB)
Collecting bottle==0.12.* (from platformio)
  Downloading https://files.pythonhosted.org/packages/ef/1a/bdc522f3e6540b73b22081797ce985ef3cb2b77143f70b506ccd9d93a83d/bottle-0.12.21-py3-none-any.whl (89kB)
Collecting click<9,>=8.0.4 (from platformio)
  Downloading https://files.pythonhosted.org/packages/c2/f1/df59e28c642d583f7dacffb1e0965d0e00b218e0186d7858ac5233dce840/click-8.1.3-py3-none-any.whl (96kB)
Collecting colorama (from platformio)
  Downloading https://files.pythonhosted.org/packages/77/8b/7550e87b2d308a1b711725dfaddc19c695f8c5fa413c640b2be01662f4e6/colorama-0.4.5-py2.py3-none-any.whl
Collecting marshmallow==3.* (from platformio)
  Downloading https://files.pythonhosted.org/packages/19/35/07a369156dc53384c381c9b2b9fa63c35478b2289b47b4ee7d81418637e5/marshmallow-3.17.0-py3-none-any.whl (48kB)
Collecting pyelftools<1,>=0.27 (from platformio)
  Downloading https://files.pythonhosted.org/packages/40/22/6706cc385a099bc5bcb0ea179c2c11863b62c79d37bd47fe464c16f9798d/pyelftools-0.28-py2.py3-none-any.whl (155kB)
Collecting pyserial==3.5.* (from platformio)
  Using cached https://files.pythonhosted.org/packages/07/bc/587a445451b253b285629263eb51c2d8e9bcea4fc97826266d186f96f558/pyserial-3.5-py2.py3-none-any.whl
Collecting requests==2.* (from platformio)
  Downloading https://files.pythonhosted.org/packages/ca/91/6d9b8ccacd0412c08820f72cebaa4f0c0441b5cda699c90f618b6f8a1b42/requests-2.28.1-py3-none-any.whl (62kB)
Collecting semantic_version==2.10.* (from platformio)
  Downloading https://files.pythonhosted.org/packages/6a/23/8146aad7d88f4fcb3a6218f41a60f6c2d4e3a72de72da1825dc7c8f7877c/semantic_version-2.10.0-py2.py3-none-any.whl
Collecting tabulate==0.8.* (from platformio)
  Downloading https://files.pythonhosted.org/packages/92/4e/e5a13fdb3e6f81ce11893523ff289870c87c8f1f289a7369fb0e9840c3bb/tabulate-0.8.10-py3-none-any.whl
Collecting zeroconf<1 (from platformio)
  Downloading https://files.pythonhosted.org/packages/c2/43/140e2fefeed540a6747020c29eebcb2a770cee474026b644bd2c4156d166/zeroconf-0.38.7-py3-none-any.whl (106kB)
Collecting aiofiles==0.8.* (from platformio)
  Downloading https://files.pythonhosted.org/packages/ca/e4/b78d049f7cc7ed053ddbfdd59b2dcc7bd387458e2c2869b602975685d65e/aiofiles-0.8.0-py3-none-any.whl
Collecting ajsonrpc==1.* (from platformio)

	at /Users/markusgschwend/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/Users/markusgschwend/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-11:
Could you remove `/Users/markusgschwend/.platformio` folder and restart VSCode. Does it work now?

phdoerfler, 2022-07-23:
I think I ran into this issue, too. I removed `.platformio`, restarted VSCode and then I got presented with:

> PlatformIO IDE installation has been suspended, because PlatformIO IDE Installer is already started in another window.

fwiw I think pip got hung up on a different package: pyserial.

I then uninstalled the Platform IO extension in VS Code, restarted, installed it again, and _now_ it seems to have finished the installation properly.

closed 2022-09-27 by stale[bot]