platformio/platformio-core-installer #2280: Installation Manager

hparaujo, 2022-07-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.68.1
  PIO IDE: v2.5.0
  System: Linux, 5.13.0-52-generic, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Linux-5.13.0-52-generic-x86_64-with-glibc2.29
Python version: 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/hparaujo/.platformio/penv
/usr/bin/python3

    at cmd (/home/hparaujo/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:187:25)
    at ChildProcess.c (/home/hparaujo/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:177:7)
    at ChildProcess.emit (node:events:390:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1064:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-11:
Could you reproduce this issue after restarting VSCode?

hparaujo, 2022-07-11:
Hi there.

Yes, every time I start VSCode, it tries to install PlatformIO and always
come up with this message.

Regards.

Hugo

ivankravets, 2022-07-15:
Could you try Local Installer Script https://docs.platformio.org/en/latest/core/installation/methods/installer-script.html#local-download-macos-linux-windows

Please provide errors here.

closed 2022-09-25 by stale[bot]