platformio/platformio-core-installer #2287: Installation Manager

hujunqiang, 2022-07-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.66.2
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.
	at u.value (c:\Users\30689\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:801864)
	at S.install (c:\Users\30689\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:4373)
	at c:\Users\30689\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27870
	at y._withProgress (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:83:43945)
	at y.withProgress (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:83:43699)
	at Object.withProgress (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:99:40824)
	at Object.startInstaller (c:\Users\30689\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:27285)
	at processTicksAndRejections (node:internal/process/task_queues:96:5)
	at async Object.activate (c:\Users\30689\.vscode\extensions\platformio.platformio-ide-2.5.0\dist\extension.js:1:22733)
  ```

ivankravets, 2022-07-15:
>  Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.

Why did you disable the built-in PlatformIO Core?

ivankravets, 2022-08-12:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2284

closed 2022-08-12 by ivankravets