platformio/platformio-core-installer #2305: Installation Manager

newbyc333, 2022-07-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.2
  PIO IDE: v2.5.0
  System: Linux, 5.15.0-40-lowlatency, x64

  %23 Exception
  ```
  Error: Traceback (most recent call last):
  File "/home/pilakkat/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 69, in <module>
    main()
  File "/home/pilakkat/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 61, in main
    bootstrap()
  File "/home/pilakkat/.platformio/.cache/tmp/get-platformio-1.1.2.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/pioinstaller/__main__.py", line 133, in main
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/click/core.py", line 1128, in __call__
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/click/core.py", line 1053, in main
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/click/core.py", line 1637, in invoke
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/click/core.py", line 1395, in invoke
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/click/core.py", line 754, in invoke
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/click/decorators.py", line 26, in new_func
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/pioinstaller/__main__.py", line 61, in cli
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/pioinstaller/core.py", line 70, in install_platformio_core
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/pioinstaller/core.py", line 98, in _install_platformio_core
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/pioinstaller/penv.py", line 71, in create_core_penv
  File "/home/pilakkat/.platformio/.cache/tmp/.piocore-installer-m3vxbh7a/tmp23lqx22f/pioinstaller.zip/pioinstaller/penv.py", line 141, in init_state
  File "/usr/lib/python3.10/subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.10/subprocess.py", line 966, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1842, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/pilakkat/.platformio/penv/bin/python'
 -> Installer version: 1.1.2
Platform: Linux-5.15.0-40-lowlatency-x86_64-with-glibc2.35
Python version: 3.10.4 (main, Jun 29 2022, 12:14:53) [GCC 11.2.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/pilakkat/.platformio/penv

	at /home/pilakkat/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5227
	at ChildProcess.c (/home/pilakkat/.vscode/extensions/platformio.platformio-ide-2.5.0/node_modules/platformio-node-helpers/dist/index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

zingo, 2022-09-18:
Seem to be the same as https://github.com/platformio/platformio-core-installer/issues/2404

zingo, 2022-09-20:
Installing python3-venv as mentioned in https://github.com/platformio/platformio-core-installer/issues/85
solved it for me on popOS! based on Ubuntu 22.04.