platformio/platformio-core-installer #2314: Installation Manager

littleanlin, 2022-07-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.2
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Configuration file contains invalid cp936 characters in C:\Users\DDD\pip\pip.ini.Error: Could not install PlatformIO Core: If you have antivirus/firewall/defender software in a system, try to disable it for a while.
 Command '['C:\\Users\\DDD\\.platformio\\penv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 2.
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.10.4 (tags/v3.10.4:9d38120, Mar 23 2022, 23:13:41) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\DDD\AppData\Local\Programs\Python\Python310\python.exe
Creating a virtual environment at C:\Users\DDD\.platformio\penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core

	at c:\Users\DDD\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5227
	at ChildProcess.c (c:\Users\DDD\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\index.js:1:5121)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\DDD\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-28:
Error: Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.

ivankravets, 2022-07-28:
Please remove this file `C:\Users\DDD\pip\pip.ini`.

closed 2022-07-28 by ivankravets