platformio/platformio-core-installer #2325: Installation Manager

zeeeeeeeeeeeoo, 2022-07-30:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.69.2
  PIO IDE: v2.5.1
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Error: Could not create PIO Core Virtual Environment. Please report to https://github.com/platformio/platformio-core-installer/issues
 -> Installer version: 1.1.2
Platform: Windows-10
Python version: 3.9.10 (tags/v3.9.10:f2f3f53, Jan 17 2022, 15:14:21) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\ZE\.platformio\python3\python.exe
Creating a virtual environment at C:\Users\ZE\.platformio\penv

	at c:\Users\ZE\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:5308
	at ChildProcess.o (c:\Users\ZE\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:4691)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-07-30:
Could you reproduce this issue after restarting VSCode?

BlackTentacle, 2022-07-30:
I just wanted to mention that I've got the same error (under Linux, however). And yet, persists even after restarting VSCode. It also persists after removing the ~/.platformio folder

zeeeeeeeeeeeoo, 2022-07-31:
Restarting didnt workIt ran perfectly after removing the ~/.platformio folder

zeeeeeeeeeeeoo, 2022-07-31:
> 

restarting didnt work
it ran perfectly after removing its folder

ivankravets, 2022-08-01:
So, do you mean that it works now?

zeeeeeeeeeeeoo, 2022-08-01:
> So, do you mean that it works now?

Sorry for my reckless judgment.It seems still does not work well.
But at least I can see the home page
I do not know if it is normal stucking in "project wizard" for hours after creating a project.
After hours,it hints an error,I can't creat projects sucessfully

ivankravets, 2022-08-01:
@zeeeeeeeeeeeoo what is your board?

ivankravets, 2022-08-01:
Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type:

```
pio pkg install -g --platform "platformio/espressif32"
```

Do you have any problems with package installation?

zeeeeeeeeeeeoo, 2022-08-01:
> Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type:
> 
> ```
> pio pkg install -g --platform "platformio/espressif32"
> ```
> 
> Do you have any problems with package installation?
it says:
PS C:\Users\ZE> pio pkg install -g --platform "platformio/espressif32"
Platform Manager: espressif32@3.4.0 is already installed
Tool Manager: toolchain-xtensa32@2.50200.97 is already installed
Tool Manager: tool-esptoolpy@1.30100.210531 is already installed

ivankravets, 2022-08-01:
It seems you have a slow internet connection and it stucks at installing arduino code. So, try the following:

```
pio project init -b esp32dev -d C:/Users/ZE/Downloads/testproject
```

Please create `testproject` folder in Downloads or change the path to the folder where you would like to init a new project. You will see all progress.

zeeeeeeeeeeeoo, 2022-08-01:
> Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type:
> 
> ```
> pio pkg install -g --platform "platformio/espressif32"
> ```
> 
> Do you have any problems with package installation?

It works well now! Just after typing the first code
Thank you so much!

ivankravets, 2022-08-01:
Could you try the following links in your browser?

1. https://dl.registry.ns2.platformio.org/tools/cb/36/aefbf6f69f5e27302916f0a1d13c03da57946e5b251be009f7cd93179d41/framework-arduinoespressif32-3.20004.0.tar.gz
2. https://hk03.layerstack.com/100MB.test

What is your download speed for 1 & 2?

zeeeeeeeeeeeoo, 2022-08-01:
> Could you try the following links in your browser?
> 
> 1. https://dl.registry.ns2.platformio.org/tools/cb/36/aefbf6f69f5e27302916f0a1d13c03da57946e5b251be009f7cd93179d41/framework-arduinoespressif32-3.20004.0.tar.gz
> 2. https://hk03.layerstack.com/100MB.test
> 
> What is your download speed for 1 & 2?

1: 10kb/s
2: 3Mb/s

ivankravets, 2022-08-01:
@zeeeeeeeeeeeoo do you use proxy or mobile network? We had a similar issue https://github.com/platformio/platformio-core/issues/4345

We hoped that finally fix the mirrors for Chinese users.

zeeeeeeeeeeeoo, 2022-08-01:
> @zeeeeeeeeeeeoo do you use proxy or mobile network? We had a similar issue [platformio/platformio-core#4345](https://github.com/platformio/platformio-core/issues/4345)
> 
> We hoped that finally fix the mirrors for Chinese users.

yes,I's using a proxy .
Thank you for your efforts for Chinese users

ivankravets, 2022-08-01:
Could you try BOTH links WITHOUT proxy? Proxy is evil.

zeeeeeeeeeeeoo, 2022-08-01:
> Could you try BOTH links WITHOUT proxy? Proxy is evil.

At the first I used the IDMDownloader with proxy and got 
1:10kb/s
2:3Mb/s

Without IDM and proxy :
1:5Mb/s
[image]
2:5Mb/s
[image]

With IDM and without proxy
1: 60kb/s
[image]
2:8Mb/s
[image]

ivankravets, 2022-08-01:
PlatformIO is open source project and is fully controlled by our community. There are no espionage components presented in russian or government-related software. Please close all these 3rd party tools, turn off the proxy and use your legal internet provider by a Chinese company. Now, open PlatformIO CLI again and change the directory to the ESP32-based project (`cd my/projectdir`). 

1. Please remove `%USERPROFILE%/.platformio/packages` directory
2. In PlatformIO Core CLI where you changed the folder to the PlatformIO project, type `pio pkg install`.

Does it download fast?

zeeeeeeeeeeeoo, 2022-08-01:
> PlatformIO is open source project and is fully controlled by our community. There are no espionage components presented in russian or government-related software. Please close all these 3rd party [...]

it seems has no response without a proxy and says "error" a few minutes later 
[image]

ivankravets, 2022-08-01:
Did you override proxy settings in VSCode? Or, set the global HTTP_PROXY environment variable? We can try without VSCode using native `cmd.exe` (Terminal), please close VSCode

```
cd C:/Users/ZE
.platformio/penv/Scripts/activate
cd Documents/PlatformIO/1
pio pkg install
```

Does it work now?

zeeeeeeeeeeeoo, 2022-08-01:
it very slow

[image]

zeeeeeeeeeeeoo, 2022-08-01:
> Did you override proxy settings in VSCode? Or, set the global HTTP_PROXY environment variable? We can try without VSCode using native `cmd.exe` (Terminal), please close VSCode
> 
> ```
> cd C:/Users/ZE
> .platformio/penv/Scripts/activate
> cd Documents/PlatformIO/1
> pio pkg install
> ```
> 
> Does it work now?

it's very slow and stuck in 0%

[image]

ivankravets, 2022-08-01:
Sorry, crazy hard to help here. You provide absolutely different results :( 

Do you mean that https://dl.registry.ns2.platformio.org/tools/cb/36/aefbf6f69f5e27302916f0a1d13c03da57946e5b251be009f7cd93179d41/framework-arduinoespressif32-3.20004.0.tar.gz still downloads well from your browser without any proxies, etc?

----

Maybe, the API is also fully blocked. Could you try the following links?

1. https://api.registry.platformio.org/v3/packages/platformio/tool/framework-arduinoespressif32?sid=4a8a9fc31dc15a4b87bb145b05db3ae0bf2333e4
2. https://api.registry.platformio.org/v3/packages/platformio/tool/framework-arduinoespressif32

The (1) MUST fail (400 error), and the second MUST be OK.

zeeeeeeeeeeeoo, 2022-08-01:
> Sorry, crazy hard to help here. You provide absolutely different results :(
>
> Do you mean that [...]

(1)
[image]
(2)
[image]

I am not very sure about what does the code excatly mean.
 it seems both （1） and （2） don't work without or with proxy.
and former link still download well.

ivankravets, 2022-08-01:
I see the root of the problem. Your IP points to Los Angeles, USA. Please remove ALL proxy settings. You still serve through the USA. As result, PlatformIO uses one of the PlatformIO's US mirrors.

zeeeeeeeeeeeoo, 2022-08-01:
> I see the root of the problem. Your IP points to Los Angeles, USA. Please remove ALL proxy settings. You still serve through the USA. As result, PlatformIO uses one of the PlatformIO's US mirrors.

I removed all my proxy settings , and ensured my IP address stay in China.
now there is a little improvement,but the download speed is still unsatisfactory

[image]

Whatever, thanks for solving my urgent need , I can  creat a ESP32 project successfully now.

ivankravets, 2022-08-01:
I've just prepared for you a special version of PlatformIO Core that will print the URL of a mirror.  How to switch to that special version?

1. Please repeat the steps mentioned in this comment https://github.com/platformio/platformio-core-installer/issues/2325#issuecomment-1201099511
2. Before running `pio pkg install`, run `pip install -U https://github.com/platformio/platformio-core/archive/refs/heads/feature/test-mirror.zip`
3. Remove some installed packages  from `~/.platformio/packages` to check the speed
4. Run `pio pkg install`

Please share here the output of `pio pkg install`.

zeeeeeeeeeeeoo, 2022-08-01:
> I've just prepared for you a special version of PlatformIO Core that will print the URL of a mirror. How to switch to that special version?
> 
> 1. Please repeat the steps mentioned in this comment [Installation Manager #2325 (comment)](https://github.com/platformio/platformio-core-installer/issues/2325#issuecomment-1201089752)
> 2. Before running `pio pkg install`, run `pip install -U https://github.com/platformio/platformio-core/archive/refs/heads/feature/test-mirror.zip`
> 3. Remove some installed packages  from `~/.platformio/packages` to check the speed
> 4. Run `pio pkg install`
> 
> Please share here the output of `pio pkg install`.

[image]
Downloading from https://dl.registry.ns2.platformio.org/tools/c1/f9/7fce33658c3efb1f2396581eac9569e693949038b99d692dba51768afbe8/toolchain-xtensa32-windows-2.50200.97.tar.gz

ivankravets, 2022-08-01:
Do you experience the same slow speed when opening the URL above in your browser?

zeeeeeeeeeeeoo, 2022-08-01:
> Do you experience the same slow speed when opening the URL above in your browser?
[image]
yes

ivankravets, 2022-08-02:
Sorry for the delay. Does https://hk03.layerstack.com/100MB.test have good downloading speed?

zeeeeeeeeeeeoo, 2022-08-03:
> Sorry for the delay. Does https://hk03.layerstack.com/100MB.test have good downloading speed?

yes it works well
[image]

closed 2022-09-27 by stale[bot]