platformio/platformio-core-installer #2344: Installation Manager

xiejin520, 2022-08-10:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.70.0
  PIO IDE: v2.5.1
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Could not find compatible PlatformIO Core. Please enable `platformio-ide.useBuiltinPIOCore` setting and restart IDE.
	at f.value (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\node_modules\platformio-node-helpers\dist\index.js:1:801940)
	at E.install (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\dist\extension.js:1:4493)
	at c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\dist\extension.js:1:27975
	at E._withProgress (d:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:83:45638)
	at E.withProgress (d:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:83:45392)
	at Object.withProgress (d:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:106:42482)
	at Object.startInstaller (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\dist\extension.js:1:27390)
	at async Object.activate (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.5.1-win32-x64\dist\extension.js:1:22838)
  ```

ivankravets, 2022-08-12:
Duplicate of https://github.com/platformio/platformio-core-installer/issues/2284

closed 2022-08-12 by ivankravets

xiejin520, 2022-08-12:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。