platformio/platformio-core-installer #2346: Installation Manager

AntoninoGratton, 2022-08-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.70.1
  PIO IDE: v2.5.1
  System: Linux, 5.13.6-arch1-1, x64

  %23 Exception
  ```
  Error:   error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      ERROR: Can not execute `setup.py` since setuptools is not available in the build environment.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error: Could not install PlatformIO Core: Command '['/home/antonino/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
 -> Installer version: 1.1.2
Platform: Linux-5.13.6-arch1-1-x86_64-with-glibc2.36
Python version: 3.10.2 (main, Jan 15 2022, 19:56:27) [GCC 11.1.0]
Python path: /usr/bin/python3
Creating a virtual environment at /home/antonino/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
PIP has been successfully updated!
Virtual environment has been successfully created!
Installing PlatformIO Core
Collecting platformio
  Using cached platformio-6.1.3.tar.gz (245 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'

	at /home/antonino/.vscode/extensions/platformio.platformio-ide-2.5.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:5308
	at ChildProcess.o (/home/antonino/.vscode/extensions/platformio.platformio-ide-2.5.1-linux-x64/node_modules/platformio-node-helpers/dist/index.js:1:4691)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-08-12:
See https://github.com/platformio/platformio-core-installer/issues/85

Did you install `apt install python3-venv` package?

AntoninoGratton, 2022-08-12:
Hi,

Yes, I already had installed the venv package.
I think upgrading the setuptools package could potentially solve the
problem, but I can't upgrade it because is installed
inside venv at the moment of the extension installation.

Greetings,

ivankravets, 2022-08-13:
What is your Linux distro?

closed 2022-09-27 by stale[bot]