platformio/platformio-core-installer #2353: Python interpreter not found

bzucak, 2022-08-12:


ivankravets, 2022-08-12:
Thanks for the report!

Please re-test with the next:

1. Close VSCode
2. Open system terminal and `cd /tmp`
3. `wget https://raw.githubusercontent.com/platformio/platformio-core-installer/develop/get-platformio.py`
4. `python3.10 get-platformio.py`
5. Start VSCode

Does it work now?

bzucak, 2022-08-13:
> Thanks for the report!
>
> Please re-test with the next:
>
>     1. Close VSCode
>
>     2. Open system terminal and `cd /tmp`
>
>     3. `wget [...]
Hello unfortunately it doesnot work.
Ive received following err:
`
Installer version: 1.1.2
Platform: Linux-4.15.0-142-generic-x86_64-with-glibc2.23
Python version: 3.10.6 (main, Aug  9 2022, 20:15:34) [GCC 5.4.0 20160609]
Python path: /usr/local/bin/python3.10
Creating a virtual environment at /home/jimi/.platformio/penv
Updating Python package manager (PIP) in a virtual environment
Virtual environment has been successfully created!
Installing PlatformIO Core
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/platformio/
WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/platformio/
WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/platformio/
WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/platformio/
WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError("Can't connect to HTTPS URL because the SSL module is not available.")': /simple/platformio/
Could not fetch URL https://pypi.org/simple/platformio/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/platformio/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
ERROR: Could not find a version that satisfies the requirement platformio (from versions: none)
ERROR: No matching distribution found for platformio
WARNING: pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping
WARNING: There was an error checking the latest version of pip.
Error: Could not install PlatformIO Core: Command '['/home/jimi/.platformio/penv/bin/python', '-m', 'pip', 'install', '-U', 'platformio']' returned non-zero exit status 1.
`

bzucak, 2022-08-13:
Hello unfortunately it doesnot work, I received the following err:
[image]

ivankravets, 2022-08-13:
It works great. Thanks for the report. The rest of issues are systems issues. Please google for the SSL error. You need to install additional apt packages

commit 028d2068 (2022-08-12): Improve support for custom Python installations // Resolve #2353